dusty-nv/jetson-inference #1246: How to convert cv::Mat to uchar3* for detectnet

Nowfal1993, 2021-10-04:
Hi. I am receiving yuv frames from the camera. which is not possbile to capture by `videoSourse::Capture()` in `detectnet.cpp` as the prefered input format are only `rgb8`, `rgba8`, `rgb32f`, `rgba32f `. In that case I am trying to capture it with `CV::VideoCapture`  But the frames captured with CV are `CV::Mat`. How Can I convert those yuv frames captured with CV to those `uchar3*` so that I can feed it to detectnet. Any help would be appreciated. Thanks in Advance

dusty-nv, 2021-10-04:
Hi @Nowfal1993, you may want to see this post, and use cvtColor() beforehand to convert YUV to RGB:

https://github.com/dusty-nv/jetson-inference/issues/129#issuecomment-324130091

Nowfal1993, 2021-10-05:
Hi @dusty-nv , I tried #129, but I am unable to perform such conversion as mentioned there. The frames are not converting properly and the programs stop immediately after 
`[TRT]    device GPU, networks/SSD-Mobilenet-v1/ssd_mobilenet_v1_coco.uff initialized`.

Query: I saw one file in the `build/arch64/include/jetson-utils/cudaYUV.h` and it seems it can be used to convert yuv422 to rgb8 or rgb32 format. Is it possoble? if so, how to integrate it to the input pipeline and to the detectnet. I stuck this point so long, Any help would be really appreciated.

dusty-nv, 2021-10-05:
> Query: I saw one file in the `build/arch64/include/jetson-utils/cudaYUV.h` and it seems it can be used to convert yuv422 to rgb8 or rgb32 format. Is it possoble?

I would recommend to use the `cudaConvertColor()` function as opposed to calling the YUV<->RGB functions directly.  See examples here:

https://github.com/dusty-nv/jetson-inference/blob/master/docs/aux-image.md#color-conversion

Nowfal1993, 2021-10-05:
```
// load the image as rgb8 (uchar3)
if( !loadImage("my_image.jpg", &imgInput, &width, &height) )
	return false;
```

Issue1: `loadImage()` function from `jetson-utils/imageIO.h` is meant to Load a  color image from disk into CUDA memory in  `ucahr3/ucahr4/float3/float4` formats with pixels values 0-255. If thats the case how can load `yuv `image of type `uint8`

closed 2023-04-27 by dusty-nv