dusty-nv/jetson-inference #1247: Can I train the posenet model?

AK51, 2021-10-06:
Hi,

The densenet121-body performs pretty well when the person is facing the camera, but some pose could not be recognized easy, e.g. the person stands at 90 degree. I tried to tuned the threshold but no luck.
And many people stand close together.
How many body model I can try? (just the default model and densenet121-body?)
Is it possible for me to train the model just like detectnet?

Thx

H19012, 2022-01-25:
How did you convert hand_pose_resnet18_baseline_att_224x224_A pth from NVIDIA-AI-IOT/trt_pose_hand to ONNX? Did you make any edits to the .pth file?

edge-ai4cv, 2023-04-22:
> How did you convert hand_pose_resnet18_baseline_att_224x224_A pth from NVIDIA-AI-IOT/trt_pose_hand to ONNX? Did you make any edits to the .pth file?

Same question here, cause I converted the hand model to ONNX following a thread in that Git but it seemed to have issues.

closed 2023-04-27 by dusty-nv