dusty-nv/jetson-inference #1249: Bounding box update and redraw

elmankarimli, 2021-10-10:
Hi,

I would like to update bounding box coordinates of stable object and redraw it after every 5 frame to somehow reduce flickering. I tried to make some changes in detectNet.cpp and detectNet.cu. Mainly in "gpuDetectionOverlayBox" and "launchDetectionOverlay" functions, like boxWidth and boxHeight(for test purposes). And did not get what I want. Basically nothing changed. 

Actually, I am new to Nvidia Nano. So, I would be very thankful for any help!

dusty-nv, 2021-10-10:
Hi, did you re-run  'make' and 'sudo make install' after making the changes?

elmankarimli, 2021-10-11:
I did run 'make' but not 'sudo make install'. I made little a 'detections[numDetections].Left = object_data[3]*width +(50)' in 'detectNet.cpp' just to see whether it is working or not. It seems worked on detection and it shifts boundix box by 50 pixels. But all I want is to take the coordinates from first frame and draw the bounding box based on those coordinates for next 4 frames.

dusty-nv, 2021-10-11:
If it is easier for you, you can skip having detectNet.cpp/.cu do the overlay (by passing overlay='none' to Detect function) and then render your own bounding boxes with openCV.