dusty-nv/jetson-inference #1274: Vision Folder - Prunning

eanmikale, 2021-11-02:
@dusty-nv - In order to Prune a model, should it be in onnx format or pth? Also, would we prune as such: python3 prunner.py ~/path-to-prunned-model? Or is there a configuration/optimization file somewhere that should point to the prunned file? 

Second, is there a README for this folder? I see this is new and want to play with the features. 

Thanks.

dusty-nv, 2021-11-02:
Hi @eanmikale, can you refer me to where in the repo you are referring to about `prunner.py`?   I don't recall it.

eanmikale, 2021-11-02:
@dusty-nv - It is an SSD submodule. See here: https://github.com/dusty-nv/pytorch-ssd/tree/8ed842a408f8c4a8812f430cf8063e0b93a56803/vision/prunning

closed 2023-04-27 by dusty-nv