dusty-nv/jetson-inference #1279: Memory Leak

rlrahulkanojia, 2021-11-05:
Hello @dusty-nv, Can you please help me with the reason for the following warning?

nvbuf_utils: dmabuf_fd 1269 mapped entry NOT found
nvbuf_utils: NvReleaseFd Failed... Exiting...
[gstreamer] gstreamer mysink taglist, video-codec=(string)"H.264\ \(Constrained\ Baseline\ Profile\)", minimum-bitrate=(uint)19420, maximum-bitrate=(uint)4039001, bitrate=(uint)332830;
nvbuf_utils: dmabuf_fd 1271 mapped entry NOT found
nvbuf_utils: NvReleaseFd Failed... Exiting...
1 objects detected
detected obj 0  class #0 (person)  confidence=0.664551
bounding box 0  (37.000000, 286.625000)  (479.000000, 359.000000)  w=442.000000  h=72.375000

detectnet is running okay and getting detections too but with the above warning. Any leads would be really helpful.

Thanks

AK51, 2022-03-22:
Hi,

I also see lots of this warning.
If I use webcam 1280x720, I don't see it, but if I use IP cam H.264 with 1920x1080/20 fps --input-rtsp-latency=0, it pops up a lots... Even on my NX...
Will it be the h.264 is too slow and timing unpredictable? But my traffic is low on my 1G switch. I can tune the input camera stream if required (like fps, constant bit rate...)
My application is time critical, is there any way, I can bypass this warning?
This warning is annoying that I cannot see my own print statement.... :<
I am looking at this forum,
https://forums.developer.nvidia.com/t/jetson-utils-videosource-capture-latency-issue/188593/16
Does it mean if I update this https://github.com/dusty-nv/jetson-utils/commit/b38357bbe33640613acb7616fd7e675adbeaab2a
I can eliminate the warning?

Thx

rlrahulkanojia, 2022-03-23:
Maybe the resolution is the key here. I fixed the warning by changing 1280 x 720 to 1280x768. 
If there is any downstream task after streaming, maybe there is a code that is accessing a pointer with a garbage value.

closed 2023-04-27 by dusty-nv