dusty-nv/jetson-inference #1281: VideoSource Capture objects do not appear to be released from memory upon deletion.

amazingproducer, 2021-11-06:
I'm trying to run predictions on the frames of a series of video frames:

```python
import jetson.utils
import jetson.inference
net = jetson.inference.detectNet("ssd-mobilenet-v2", threshold=0.5)
video_file = jetson.utils.videoSource("video.mp4")
video_file.Open()
while video_file.IsStreaming():
    img = video_file.Capture()
    print(f"Memory will be flushed upon object deletion: {img.freeOnDelete}")
    break
video_file.Close()
```

> `Memory will be flushed upon object deletion: False`

As a result of the above, the Jetson memory simply fills up when using it with iterators. Moreover, it appears as though the `freeOnDelete` `cudaMemory` attribute cannot be revised. By what means are python users expected to free the cuda memory after allocating it?

dusty-nv, 2021-11-08:
Hi @amazingproducer, the frames from `videoSource.Capture()` are flagged like that with `freeOnDelete=false` because internally the videoSource implementation uses a ring buffer where it re-uses the previous memory buffers.

I'm curious, if you use the `detectnet.py` sample instead, does the memory fill up also?   If not, you may want to base your code on that sample instead.

amazingproducer, 2021-11-09:
As it turns out, my memory issues had more to do with the way I was using `GetClassDesc` for each detection. Memory is freed as expected when the offending line is omitted.

closed 2021-11-09 by amazingproducer