dusty-nv/jetson-inference #1282: Can I build project from source for a Gefore dGPU machine?

DivyankaThakur03, 2021-11-06:
Hi @dusty-nv , Thanks for the awesome tutorials. I wanted to know whether I can build my project from source for a dGPU device as I don't own a Jetson hardware.
I tried cloning the repo but I am facing issue with the `make -j$(nproc)` command.
Error message- ```
~/jetson-inference/build$ make -j$(nproc)
make: *** No targets specified and no makefile found.  Stop.
```

[I own a GTX 1650 card]
Thanks

dusty-nv, 2021-11-08:
Did you run `cmake ../` beforehand?

Building this project on x86 isn't officially supported, and there are likely to be errors that you would need to workaround.  However some people have done it.

DivyankaThakur03, 2021-11-10:
```
The system is: Linux - 5.4.0-89-generic - x86_64
Compiling the C compiler identification source file "CMakeCCompilerId.c" succeeded.
Compiler: /usr/bin/cc 
Build flags: 
Id flags:  

The output was:
0

Compilation of the C compiler identification source "CMakeCCompilerId.c" produced "a.out"

The C compiler identification is GNU, found in "/home/divya/jetson-inference/build/CMakeFiles/3.10.2/CompilerIdC/a.out"

Compiling the CXX compiler identification source file "CMakeCXXCompilerId.cpp" succeeded.
Compiler: /usr/bin/c++ 
Build flags: 
Id flags:  

The output was:
0

Compilation of the CXX compiler identification source "CMakeCXXCompilerId.cpp" produced "a.out"

The CXX compiler identification is GNU, found in "/home/divya/jetson-inference/build/CMakeFiles/3.10.2/CompilerIdCXX/a.out"

Determining if the C compiler works passed with the following output:
Change Dir: /home/divya/jetson-inference/build/CMakeFiles/CMakeTmp

Run Build Command:"/usr/bin/make" "cmTC_8a8de/fast"
/usr/bin/make -f CMakeFiles/cmTC_8a8de.dir/build.make CMakeFiles/cmTC_8a8de.dir/build
make[1]: Entering directory '/home/divya/jetson-inference/build/CMakeFiles/CMakeTmp'
Building C object CMakeFiles/cmTC_8a8de.dir/testCCompiler.c.o
/usr/bin/cc    -o CMakeFiles/cmTC_8a8de.dir/testCCompiler.c.o   -c /home/divya/jetson-inference/build/CMakeFiles/CMakeTmp/testCCompiler.c
Linking C executable cmTC_8a8de
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_8a8de.dir/link.txt --verbose=1
/usr/bin/cc      -rdynamic CMakeFiles/cmTC_8a8de.dir/testCCompiler.c.o  -o cmTC_8a8de 
make[1]: Leaving directory '/home/divya/jetson-inference/build/CMakeFiles/CMakeTmp'

Detecting C compiler ABI info compiled with the following output:
Change Dir: /home/divya/jetson-inference/build/CMakeFiles/CMakeTmp

Run Build Command:"/usr/bin/make" "cmTC_92efd/fast"
/usr/bin/make -f CMakeFiles/cmTC_92efd.dir/build.make CMakeFiles/cmTC_92efd.dir/build
make[1]: Entering directory '/home/divya/jetson-inference/build/CMakeFiles/CMakeTmp'
Building C object CMakeFiles/cmTC_92efd.dir/CMakeCCompilerABI.c.o
/usr/bin/cc    -o CMakeFiles/cmTC_92efd.dir/CMakeCCompilerABI.c.o   -c /usr/share/cmake-3.10/Modules/CMakeCCompilerABI.c
Linking C executable cmTC_92efd
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_92efd.dir/link.txt --verbose=1
/usr/bin/cc     -v -rdynamic CMakeFiles/cmTC_92efd.dir/CMakeCCompilerABI.c.o  -o cmTC_92efd 
Using built-in specs.
COLLECT_GCC=/usr/bin/cc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-linux-gnu/7/lto-wrapper
OFFLOAD_TARGET_NAMES=nvptx-none
OFFLOAD_TARGET_DEFAULT=1
Target: x86_64-linux-gnu
Configured with: ../src/configure -v --with-pkgversion='Ubuntu 7.5.0-3ubuntu1~18.04' --with-bugurl=file:///usr/share/doc/gcc-7/README.Bugs --enable-languages=c,ada,c++,go,brig,d,fortran,objc,obj-c++ --prefix=/usr --with-gcc-major-version-only --program-suffix=-7 --program-prefix=x86_64-linux-gnu- --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --libdir=/usr/lib --enable-nls --enable-bootstrap --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --with-default-libstdcxx-abi=new --enable-gnu-unique-object --disable-vtable-verify --enable-libmpx --enable-plugin --enable-default-pie --with-system-zlib --with-target-system-zlib --enable-objc-gc=auto --enable-multiarch --disable-werror --with-arch-32=i686 --with-abi=m64 --with-multilib-list=m32,m64,mx32 --enable-multilib --with-tune=generic --enable-offload-targets=nvptx-none --without-cuda-driver --enable-checking=release --build=x86_64-linux-gnu --host=x86_64-linux-gnu --target=x86_64-linux-gnu
Thread model: posix
gcc version 7.5.0 (Ubuntu 7.5.0-3ubuntu1~18.04) 
COMPILER_PATH=/usr/lib/gcc/x86_64-linux-gnu/7/:/usr/lib/gcc/x86_64-linux-gnu/7/:/usr/lib/gcc/x86_64-linux-gnu/:/usr/lib/gcc/x86_64-linux-gnu/7/:/usr/lib/gcc/x86_64-linux-gnu/
LIBRARY_PATH=/usr/lib/gcc/x86_64-linux-gnu/7/:/usr/lib/gcc/x86_64-linux-gnu/7/../../../x86_64-linux-gnu/:/usr/lib/gcc/x86_64-linux-gnu/7/../../../../lib/:/lib/x86_64-linux-gnu/:/lib/../lib/:/usr/lib/x86_64-linux-gnu/:/usr/lib/../lib/:/usr/lib/gcc/x86_64-linux-gnu/7/../../../:/lib/:/usr/lib/
COLLECT_GCC_OPTIONS='-v' '-rdynamic' '-o' 'cmTC_92efd' '-mtune=generic' '-march=x86-64'
 /usr/lib/gcc/x86_64-linux-gnu/7/collect2 -plugin /usr/lib/gcc/x86_64-linux-gnu/7/liblto_plugin.so -plugin-opt=/usr/lib/gcc/x86_64-linux-gnu/7/lto-wrapper -plugin-opt=-fresolution=/tmp/ccm0ngBU.res -plugin-opt=-pass-through=-lgcc -plugin-opt=-pass-through=-lgcc_s -plugin-opt=-pass-through=-lc -plugin-opt=-pass-through=-lgcc -plugin-opt=-pass-through=-lgcc_s --build-id --eh-frame-hdr -m elf_x86_64 --hash-style=gnu --as-needed -export-dynamic -dynamic-linker /lib64/ld-linux-x86-64.so.2 -pie -z now -z relro -o cmTC_92efd /usr/lib/gcc/x86_64-linux-gnu/7/../../../x86_64-linux-gnu/Scrt1.o /usr/lib/gcc/x86_64-linux-gnu/7/../../../x86_64-linux-gnu/crti.o /usr/lib/gcc/x86_64-linux-gnu/7/crtbeginS.o -L/usr/lib/gcc/x86_64-linux-gnu/7 -L/usr/lib/gcc/x86_64-linux-gnu/7/../../../x86_64-linux-gnu -L/usr/lib/gcc/x86_64-linux-gnu/7/../../../../lib -L/lib/x86_64-linux-gnu -L/lib/../lib -L/usr/lib/x86_64-linux-gnu -L/usr/lib/../lib -L/usr/lib/gcc/x86_64-linux-gnu/7/../../.. CMakeFiles/cmTC_92efd.dir/CMakeCCompilerABI.c.o -lgcc --push-state --as-needed -lgcc_s --pop-state -lc -lgcc --push-state --as-needed -lgcc_s --pop-state /usr/lib/gcc/x86_64-linux-gnu/7/crtendS.o /usr/lib/gcc/x86_64-linux-gnu/7/../../../x86_64-linux-gnu/crtn.o
[... cut ...]
    Feature record: CXX_FEATURE:0cxx_thread_local
    Feature record: CXX_FEATURE:0cxx_trailing_return_types
    Feature record: CXX_FEATURE:0cxx_unicode_literals
    Feature record: CXX_FEATURE:0cxx_uniform_initialization
    Feature record: CXX_FEATURE:0cxx_unrestricted_unions
    Feature record: CXX_FEATURE:0cxx_user_literals
    Feature record: CXX_FEATURE:0cxx_variable_templates
    Feature record: CXX_FEATURE:0cxx_variadic_macros
    Feature record: CXX_FEATURE:0cxx_variadic_templates
```

cmake log

rajeshroy402, 2021-12-07:
Is there a solution to this? @dusty-nv

dusty-nv, 2021-12-07:
Hi @rajeshroy402, unfortunately I don't know, as I don't build/support this project for x86 (only for Jetson / aarch64)

In theory it should be possible but there are some Jetson-specific features and libraries used.

closed 2023-04-27 by dusty-nv