dusty-nv/jetson-inference #1283: docker/run.sh returns error

detruby, 2021-11-06:
after some updates applied, running
docker/run.sh
returns
docker: Error response from daemon: failed to create shim: OCI runtime create failed: container_linux.go:380: starting container process caused: error adding seccomp filter rule for syscall clone3: permission denied: unknown.

rlrahulkanojia, 2021-11-06:
Did you try this using sudo permissions?

albcaraveo, 2021-11-07:
Hello, I have the same issue :(

detruby, 2021-11-07:
I was able to recreate the cause. I re imaged the SDcard with
jetson-nano-jp46-sd-card-image.zip
booted, etc
git clone --recursive https://github.com/dusty-nv/jetson-inference
cd jetson-inference
docker/run.sh
setup models, etc
exit
docker/run.sh
works as expected

exit
cd ~
sudo apt update
sudo apt-get update
cd jetson-inference
docker/run.sh
works as expected

exit
cd ~
sudo apt-get upgrade
the above upgrade docker and a bunch of things
cd jetson-inference
docker/run.sh
fails

carsten12, 2021-11-07:
Same here on Jetson Nano 2GB with nvcr.io and own docker images.

hessio, 2021-11-07:
So @detruby are you saying that when you reimaged the sd card it started to work for you then?

carsten12, 2021-11-07:
There is a more easy way: downgrade docker back to 19 and containderd and pin it. I send a link to a description when I am back at home.

carsten12, 2021-11-07:
So here it is: https://github.com/dusty-nv/jetson-containers/issues/108#issuecomment-962597353

The problem is related to a upgrade towards docker 20 for some of the latest ubuntu images due to security reasons. It seems to break using the nvcr.io images. I guess the maintainer is going to fix that, but in the meantime you can download the docker 19 .deb file and install it as well as the related containerd version. Pin the versions as described to prevent them from being upgraded next time you call apt-get upgrade. All nvrc.io and my own images are now working again.

Robotmad, 2021-11-07:
@dusty-nv - this is going to hit so many people who will each individually spend ages trying to fix it until they come across this work around as the assumption is normally that you've done something wrong, hence start from scratch with a new SD card image etc....  I appreciate it is GTC this week but I hope Nvidia are going to sort this properly as a matter of urgency - perhaps worth having a pinned announcement in the forums to get people to pin the versions in the mean time?

hessio, 2021-11-08:
Oh thanks for the update @carsten12, the work around I used was to just install and build from source it's still really painless and quick to do.

dusty-nv, 2021-11-08:
Thanks everyone for reporting this over the weekend and trying/confirming the various workarounds!  The engineers are working to release an updated package as soon as possible.

PR #1288 (2021-11-11): Resolves seccomp error
Closes #1283.
Disables seccomp error for updated versions of docker.
closed 2021-11-11 by dusty-nv