dusty-nv/jetson-inference #1284: failed to handle incoming buffer,  map buffer size was less than max size

freezinghands, 2021-11-08:
Hi, 

I have jetson nano and followed all instructions at 'Hello AI World' by 'Building the Project from Source' without using docker container.

At location <jetson-inference>/build/aarch64/bin I tried '$ ./imagenet --network=resnet-18 ~/cuda/HW3/Video.mp4 ~/cuda/HW3/Video_out.mp4' in terminal and waited for inference. 

Estimated output did not come out, and below messages printed endlessly in my terminal.

`
[gstreamer] gstBufferManager -- failed to map EGLImage from NVMM buffer
`
`
[gstreamer] gstDecoder -- failed to handle incoming buffer
`
`
[gstreamer] gstBufferManager -- map buffer size was less than max size (1008 vs 3110400)
`

I tried image recognition by typing '$ ./imagenet ~/cuda/HW3/Image4.jpg ~/cuda/HW3/Image4_output_2.jpg' like this and error did not occur.

Same error message printed when i used CSI camera, by typing '$ ./imagenet csi://0'. 
Also same error message by typing '$ video-viewer csi://0'

Hardware is not problem and i tested few hours ago in other board.

How can i solve this problem?

Thanks.

dusty-nv, 2021-11-08:
Hmm... can you try running this @freezinghands?

``` bash
cd /path/to/your/jetson-inference/build
cmake -DDISABLE_NVMM ../
make
sudo make install
```

freezinghands, 2021-11-09:
I got solution from branch 'nvmm-disabled' branch.
Errors gone when I did this. 
Thanks for help.

closed 2021-11-09 by freezinghands

dusty-nv, 2023-10-26:
@changx03 does it happen if you build with `-DENABLE_NVMM=off` during cmake ?