dusty-nv/jetson-inference #1285: Using inference from a virtual environment

LewsTherin511, 2021-11-08:
Hi, I'm trying to build and run the project from a virtual environment (I tried both with conda and virtualenv, the results are the same) 
I ran the whole building procedure successfully, and when I ran 'cmake ../' the process even found the Python3.6 version in the conda virtual environment:

```
 detecting Python 3.6...
-- found Python version:  3.6 (3.6.13)
-- found Python include:  /home/lews/archiconda3/envs/tf/include/python3.6m
-- found Python library:  /home/lews/archiconda3/envs/tf/lib/libpython3.6m.so
-- CMake module path:  /home/lews/projects/jetson-inference/utils/cuda;/home/lews/projects/jetson-inference/python/bindings;/home/lews/projects/jetson-inference/python/bindings/../../utils/python/bindings
-- NumPy ver. 1.19.5 found (include: /home/lews/archiconda3/envs/tf/lib/python3.6/site-packages/numpy/core/include)
-- found NumPy version:  1.19.5
-- found NumPy include:  /home/lews/archiconda3/envs/tf/lib/python3.6/site-packages/numpy/core/include

```

I can use detectnet from the system python installation, but when switching to another virtual environment I get the error

    "ModuleNotFoundError: No module named 'jetson'"

Is there a way to do this, or should I just avoid environments?

dusty-nv, 2021-11-08:
I haven't used virtual environments with this project before, so I'm of limited help with this, sorry about that.

However after installing jetson-inference, can you make the virtualenv with `virtualenv --system-site-packages` ?

nikunjlad, 2022-07-27:
@dusty-nv @LewsTherin511
I could get the **jetson.inference** and **jetson.utils** working inside my virtual environment.  As correctly mentioned in the build procedure, the installation happens under **/usr/lib/pythonXX/dist-packages** location. The key is not just copy the .so files from this directory to site-packages inside virtual environment but also 2 other folders namely the **jetson** and **Jetson** folder. Do this and it should work -

`cp /usr/lib/pythonX.X/dist-packages/jetson_utils_python.so <your_virtual_env>/lib/pythonX.X/site-packages`
`cp /usr/lib/pythonX.X/dist-packages/jetson_inference_python.so <your_virtual_env>/lib/pythonX.X/site-packages`
`cp -r /usr/lib/pythonX.X/dist-packages/jetson <your_virtual_env>/lib/pythonX.X/site-packages`
`cp -r /usr/lib/pythonX.X/dist-packages/Jetson <your_virtual_env>/lib/pythonX.X/site-packages`

Once you move both the .so files and the 2 Jetson folders inside the virtual environment, activate your virtual environment and try importing **jetson.inference** and **jetson.utils**. They should work well. Attaching a SS for reference:

[image]

closed 2023-04-27 by dusty-nv