dusty-nv/jetson-inference #1289: libjetson-inference.so: undefined reference to `cublasDestroy_v2'

kk52099, 2021-11-10:
hi, @dusty-nv  ，can u give me some suggestion? 
[image]

dusty-nv, 2021-11-10:
@kk52099 what version of JetPack-L4T is running on your Jetson?

kk52099, 2021-11-11:
> @kk52099 what version of JetPack-L4T is running on your Jetson?

@dusty-nv L4t-pytorch,  i  only run the "cmke  ../    &&make" in my docker images , SO When the project will be builded,  I will   copy the bin file to  my NX board  and run it

dusty-nv, 2021-11-11:
How did you start the container?  Was it using one of my docker/run.sh scripts, or if not was the container started with `--runtime nvidia`?

If this is from when you are building a Dockerfile, set your default docker runtime to `nvidia`, like shown here:  https://github.com/dusty-nv/jetson-containers#docker-default-runtime

kk52099, 2021-11-11:
@dusty-nv  you are right ,  the container was not started with --runtime nvidia,   because i run the container on x86 amd

dusty-nv, 2021-11-11:
Ah, right.  Change the base container to one of the `nvcr.io/nvidia/pytorch` containers from https://catalog.ngc.nvidia.com/orgs/nvidia/containers/pytorch (these are for x86)
l4t-pytorch is for Jetson only.

Then run it like shown on that NGC page (i.e. with `--gpus all`)
You have to have the NVIDIA driver and NVIDIA container runtime installed on your x86 machine.

kk52099, 2021-11-11:
@dusty-nv  it have not GPU,only CPU,    then  i start the contain by the command "sudo docker run -it --net host nvcr.io/nvidia/l4t-pytorch:r32.4.3-pth1.6-py3",  it is  OK to  build  the tensorrt sample(sampleSSD) ,  but i can not  success to build your project

dusty-nv, 2021-11-11:
If it doesn't have GPU, it's not going to run jetson-inference anyways (which uses GPU and CUDA).  You may have other build errors with jetson-inference as well because I build it on Jetson and not x86.

kk52099, 2021-11-11:
@dusty-nv Ok, thinks very much , then  i  run the command "docker/build.sh",  the info happen
[image], why?

closed 2023-04-27 by dusty-nv