dusty-nv/jetson-inference #1290: libcudart.so, libcublas.so do not exist in container dustynv/jetson-inference:r32.4.3

michaelblindnology, 2021-11-10:
Hi,
I build my image based on  `dustynv/jetson-inference:r32.4.3`.
I need to compile `opencv 3.4.14` from sources.
However libcudart not exist.
/usr/lib/aarch64-linux-gnu/libcublas.so  file not recognized: File truncated
As you can see:
```
root@2705d68d0852:/# ll /usr/lib/aarch64-linux-gnu/libcublas.so
lrwxrwxrwx 1 root root 15 Oct 27  2020 /usr/lib/aarch64-linux-gnu/libcublas.so -> libcublas.so.10
root@2705d68d0852:/# ll /usr/lib/aarch64-linux-gnu/libcublas.so.10
libcublas.so.10         libcublas.so.10.2.2.89  
root@2705d68d0852:/# ll /usr/lib/aarch64-linux-gnu/libcublas.so.10
lrwxrwxrwx 1 root root 22 Oct 27  2020 /usr/lib/aarch64-linux-gnu/libcublas.so.10 -> libcublas.so.10.2.2.89
root@2705d68d0852:/# ll /usr/lib/aarch64-linux-gnu/libcublas.so.10.2.2.89 
-rw-r--r-- 1 root root 0 Jul  1  2020 /usr/lib/aarch64-linux-gnu/libcublas.so.10.2.2.89
root@2705d68d0852:/# file /usr/lib/aarch64-linux-gnu/libcublas.so.10.2.2.89 
**/usr/lib/aarch64-linux-gnu/libcublas.so.10.2.2.89: empty**
```
Please help,

Thanks.

dusty-nv, 2021-11-10:
Hi @michaelblindnology, please try setting your default docker runtime to `nvidia` as shown here:

https://github.com/dusty-nv/jetson-containers#docker-default-runtime

This way, the nvidia runtime will be used during `docker build` operations.

michaelblindnology, 2021-11-10:
Hi,
I added the `nvidia-container-runtime` to daemon.json
When I run :
```
> Executing task: docker run --rm -it  ssh-test:latest <

WARNING: The requested image's platform (linux/arm64) does not match the detected host platform (linux/amd64) and no specific platform was requested
docker: Error response from daemon: OCI runtime create failed: unable to retrieve OCI runtime error (open /run/containerd/io.containerd.runtime.v2.task/moby/50226fb223da8cbe3003aacdce6fbe50bf6e190dad7db47b48a0648c9b17c9b4/log.json: no such file or directory): exec: "nvidia-container-runtime": executable file not found in $PATH: <nil>: unknown.
The terminal process "bash '-c', 'docker run --rm -it  ssh-test:latest'" terminated with exit code: 127.

Terminal will be reused by tasks, press any key to close it.
```

dusty-nv, 2021-11-10:
Are you running this on x86?   The jetson-inference container is for arm64.

michaelblindnology, 2021-11-10:
I know. I'm running on x86.
But when I run 
`docker run  --runtime nvidia --rm --privileged multiarch/qemu-user-static --reset -p yes`
I receive:
`docker: Error response from daemon: OCI runtime create failed: unable to retrieve OCI runtime error (open /run/containerd/io.containerd.runtime.v2.task/moby/27fcf2af997545b491d54176e7d54fcde956e2a76a3fa6c9257f7eb5388fd450/log.json: no such file or directory): exec: "nvidia-container-runtime": executable file not found in $PATH: unknown.`

When I'm running different container
`FROM nvcr.io/nvidia/l4t-cuda:10.2.460-runtime`
I have **cuda** but not **TensorRT**.

Thanks.
Michael.

kk52099, 2021-11-11:
@dusty-nv ok,thanks,  now  i run the command"docker/build.sh",why happen the info?  
[image]

kk52099, 2021-11-11:
> I know. I'm running on x86. But when I run `docker run --runtime nvidia --rm --privileged multiarch/qemu-user-static --reset -p yes` I receive: `docker: Error response from daemon: OCI runtime create failed: unable to retrieve OCI runtime error (open /run/containerd/io.containerd.runtime.v2.task/moby/27fcf2af997545b491d54176e7d54fcde956e2a76a3fa6c9257f7eb5388fd450/log.json: no such file or directory): exec: "nvidia-container-runtime": executable file not found in $PATH: unknown.`
> 
> When I'm running different container `FROM nvcr.io/nvidia/l4t-cuda:10.2.460-runtime` I don't receive this error. But I have **cuda** but not **TensorRT** and I need both.
> 
> Thanks. Michael.
is the l4t-cuda:10.2.460-runtime  for x86?

michaelblindnology, 2021-11-11:
Hi,
I think I need to clarify.
I want to build on x86 machine, container that supports Jetson Xavier + cuda + TensorRT + ROS melodic.
I want to run this container on Jetson Xavier(arm64).
Which container should I use ?

Thanks,
Michael.

michaelblindnology, 2021-11-11:
> I know. I'm running on x86. But when I run `docker run --runtime nvidia --rm --privileged multiarch/qemu-user-static --reset -p yes` I receive: `docker: Error response from daemon: OCI runtime [...]

>> to build on x86 and run on arm64.

dusty-nv, 2021-11-11:
> I want to build on `x86 `machine, container that supports `Jetson Xavier` +`cuda`+ `TensorRT `+ `ROS melodic`.
> I want to run this container on Jetson Xavier(`arm64`).
> Which container should I use ?

I'm not sure, I only build containers on Jetson.

You can check with @rbonghi , he has built some arm64/Jetson containers on x86.

michaelblindnology, 2021-11-11:
Thanks,
If I want to build on Jetson, Which container should I use for `Jetson Xavie`r +`cuda`+ TensorRT + ROS melodic ?

closed 2023-04-27 by dusty-nv