dusty-nv/jetson-inference #1291: 100 GB Swapfile on 256GB Micro-SD not enough?

eanmikale, 2021-11-11:
Dusty, 
Thank you, you've been great. We are creating a pipeline using jetson-inference, however, even with 100Gb of swap space on a 256GB Micro-SD card, we have only once successfully trained a cat_dog model using the /jetson-inference/python/training/classification/train.py file and directory. It either hangs, is killed, with the following code: 

python3 train.py --epochs=1 --model-dir=models/cat_dog data/cat_dog

Attached is a screenshot of our Gparted Analysis and memory distribution. 
[image]

dusty-nv, 2021-11-11:
Not everything can go into swap (like GPU memory), so there is typically an upper limit to what is needed/effective when it comes to swap.  I think the most I ever mounted was 16GB (on an AGX Xavier)

Which Jetson are you using?
Have you tried disabling the GUI like here:  https://github.com/dusty-nv/jetson-inference/blob/master/docs/pytorch-transfer-learning.md#disabling-the-desktop-gui

And also running train.py with `--batch-size=1 --workers=0` like mentioned here:  https://github.com/dusty-nv/jetson-inference/blob/master/docs/pytorch-cat-dog.md#re-training-resnet-18-model

eanmikale, 2021-11-11:
Dusty,Great to know. We have disabled the GUI and set workers=0. If we sudo init 3 in a script, how can we bypass the username, password prompt to go straight into training? Any ideas? We are setting up a desktop application to allow us to train in one-click, for non-developers. We are using the Nano for prototyping. We have an Xavier, but we have 8 production A.I. Apps on the Xavier, as we created an A.I. Marketplace-on-a-Chip. And unfortunately the 16GB of extra RAM is a bottleneck on the Xavier, which brought us to the Nano. But still having issues with memory even when we clear the cache, and follow your suggestions.  Can we trick the memory into thinking there is more?  ---- On Wed, 10 Nov 2021 20:22:27 -0600  ***@***.***  wrote ----
Not everything can go into swap (like GPU memory), so there is typically an upper limit to what is needed/effective when it comes to swap.  I think the most I ever mounted was 16GB (on an AGX Xavier)
Which Jetson are you using?
Have you tried disabling the GUI like here:  https://github.com/dusty-nv/jetson-inference/blob/master/docs/pytorch-transfer-learning.md#disabling-the-desktop-gui
And also running train.py with --batch-size=1 --workers=0 like mentioned here:  https://github.com/dusty-nv/jetson-inference/blob/master/docs/pytorch-cat-dog.md#re-training-resnet-18-model

dusty-nv, 2021-11-11:
> how can we bypass the username, password prompt to go straight into training?

I'm not sure.  Usually I just SSH into the board at that point.

eanmikale, 2021-11-11:
Ok. Thank you. We just may write a boot up script to run at boot. Here a link for others looking to run a script on the Jetsons at boot: https://askubuntu.com/questions/290099/how-to-run-a-script-during-boot-as-root. ---- On Thu, 11 Nov 2021 08:01:28 -0600  ***@***.***  wrote ----

how can we bypass the username, password prompt to go straight into training?

I'm not sure.  Usually I just SSH into the board at that point.

eanmikale, 2021-11-11:
You can close this.  ---- On Thu, 11 Nov 2021 08:01:28 -0600  ***@***.***  wrote ----

how can we bypass the username, password prompt to go straight into training?

I'm not sure.  Usually I just SSH into the board at that point.

closed 2021-11-11 by dusty-nv