dusty-nv/jetson-inference #1294: detect vehicles, peoples and a custom object from camera

shaojun, 2021-11-11:
hi dusty, thanks for the great repo.
I'm using **Jetson nano 2G**, and it runs well based on your repo, I'm planning use it in **production**, and I have 4 classes of objects need to detecting on the scene:

- Bicycle
- Motor cycle
- people
- A custom door sign

I was told for production, should always use models from [nvidia ngc](https://catalog.ngc.nvidia.com/orgs/nvidia/teams/tao/models), from my check, there's no single model can santisfies, i can see
[peoplenet](https://catalog.ngc.nvidia.com/orgs/nvidia/teams/tao/models/peoplenet) for detect people, and [trafficcamnet](https://catalog.ngc.nvidia.com/orgs/nvidia/teams/tao/models/trafficcamnet) for car and bicycle (**but no motocycle**).

For the custom door sign detection, I understand i have no way but to use transfer learning(the pre-trained model also should get from ngc) to train a new model which **only** for detect the door sign.

So could you give some advise, what is the best way for my situation, mix different models(does my device performance can support this way?) or just directly using the one from your tutorial (like model trained on MS COCO)?

dusty-nv, 2021-11-11:
Hi @shaojun, you are correct that the NGC models have superior accuracy and performance (and you would run those through DeepStream SDK).  If you use those, I would also recommend getting a training setup (e.g. a cloud instance if you don't have an x86 machine with GPU)

In the meantime, if the MS COCO model is working fine for you, then you can stick with that for now.   Due to the performance/memory of Nano 2GB, I would recommend trying to use just one model if possible.

shaojun, 2021-11-11:
thanks for the reply.

so i don't have too much choice as the limited performance of the nano 2g, if i understand correct, single model is the only way now,  need download the MS COCO dataset plus my custom door sign dataset, then train a model from scratch?

dusty-nv, 2021-11-11:
I would pick only the object classes that you need, because the fewer classes the model has the faster it will be (because the complexity of the model is lower)

I believe the Pascal VOC dataset supports bicycle + motorcycle + people, and you could in theory add your custom class to it.  It will require some manipulation of the dataset to filter out the classes you don't want.  

If you are training using pytorch-ssd and train_ssd.py from this project, the challenge is getting the dataset all into the same format.  I typically use Pascal VOC format because it's easier to work with and make custom data for.

shaojun, 2021-11-12:
thanks dusty.
Today I take a brief look at `deep stream 6` and installed it by docker(with the docker image contains samples in) , feels more complicated especially for those config files.

Just wonder, for my camera detection scenario(for production), can I just go with your repo's docker and [detectnet](https://github.com/dusty-nv/jetson-inference/blob/master/python/examples/detectnet.py) since it looks much clearer.

dusty-nv, 2021-11-12:
> can I just go with your repo's docker and [detectnet](https://github.com/dusty-nv/jetson-inference/blob/master/python/examples/detectnet.py) since it looks much clearer.

Technically you can yes, if it meets your performance needs.  DeepStream will ultimately give you better performance.

shaojun, 2021-11-18:
thanks Dusty.
By train with your steps, finally we will convert the model with `.pth` to `.onnx` and then put into deepstream, what is the different to train dataset directly by [TAO](https://developer.nvidia.com/tao-toolkit)? And I see TAO even not support `VOC` format.

dusty-nv, 2021-11-18:
You can run the pytorch-ssd ssd-mobilenet ONNX through DeepStream with the following resources:

* https://elinux.org/index.php?title=Jetson/L4T/TRT_Customized_Example#Custom_Parser_for_SSD-MobileNet_Trained_by_Jetson-inference
* https://forums.developer.nvidia.com/t/deepstream-5-1-pytorch-mobilenet-ssd-v1-retained-onnx-poor-performance/175468

TAO can give you better performance because it can prune the model.  TAO would run on an x86 Linux system with NVIDIA GPU

shaojun, 2021-11-19:
thanks dusty.
the first link is for all `SSD` related models, correct? that means VGG16-ssd should also be applied with the steps for run with deepstream.

closed 2023-04-27 by dusty-nv