dusty-nv/jetson-inference #1295: Posenet - Failed to load topology json - Pose-ResNet18

kelguerin, 2021-11-11:
Hi Dustin,

First, thanks for putting together this awesome library!

I am trying the posenet example as documented here:
[here](https://github.com/dusty-nv/jetson-inference/blob/master/docs/posenet.md#pre-trained-pose-estimation-models) with the python example. 

When I run `$ ./posenet.py --network=resnet18-hand /dev/video0` I get the following issue:

```bash
poseNet -- loading pose estimation model from:
        -- model        networks/Pose-ResNet18-Hand/pose_resnet18_hand.onnx
        -- topology     networks/Pose-ResNet18-Hand/hand_pose.json
        -- colors       networks/Pose-ResNet18-Hand/colors.txt
        -- input_blob   'input'
        -- output_cmap  'cmap'
        -- output_paf   'paf'
        -- threshold    0.150000
        -- batch_size   1

[TRT]    poseNet -- failed to load topology json from 'networks/Pose-ResNet18-Hand/hand_pose.json'
```
I am running this from inside the container. I have downloaded ResNet-18 from [here](https://github.com/dusty-nv/jetson-inference/releases) per the instructions for if the model downloader fails.  Looking at the contents of that file (ResNet-18) there does not seem to be any of the files posenet is looking for, such as the `pose_resnet18_hand.onnx` but rather a caffemodel instead. 

```bash
kel@jetson-nano:~/.../networks$ tree ResNet-18
ResNet-18
├── deploy.prototxt
├── ResNet-18.caffemodel
├── ResNet_mean.binaryproto
├── solver.prototxt
└── train.prototxt

0 directories, 5 files

```

I'm not sure where else to look for those models.

Thanks in advance for the help!
Best,
Kel

dusty-nv, 2021-11-11:
Do you have the `jetson-inference/data/networks/Pose-ResNet18-Hand/` directory?

``` bash
ls -ll data/networks/Pose-ResNet18-Hand/
total 81128
-rw-rw-r-- 1 nvidia nvidia      217 Jul 29 17:58 colors.txt
-rw-rw-r-- 1 nvidia nvidia      596 Jul 26 18:28 hand_pose.json
-rw-rw-r-- 1 nvidia nvidia 83064271 Jul 26 18:27 pose_resnet18_hand.onnx
```

kelguerin, 2021-11-11:
No  I don't. When I ran `model_downloader.sh` I see this error, which lead me to try and download it manually.

[image]

kelguerin, 2021-11-11:
Ok so I went ahead and just downloaded those two files directly from the address in the script, and put them in my directory. In order to use the command from the tutorial, I had to run the command from the data directory, because it seems that it's looking for the network relative to where the command is executed.

I think this will work for me for now, but I'm not sure why the downloader doesn't seem to work for me.

zubairahmed-ai, 2021-11-15:
Hi @dusty-nv I am also facing this and within the docker container, I even copied these two folders within my working directory from the `jetson-inference` like @kelguerin but still no joy.

zubairahmed-ai, 2021-11-15:
Just so you know I am also using `net = jetson.inference.detectNet("ssd-mobilenet-v2", threshold=0.5)` in another script and this loads fine after some initial docker volume mount issues in my docker container but within the same container I am unable to load `net = jetson.inference.poseNet("resnet18-body", None, threshold=0.15)` and this results in topology error

zubairahmed-ai, 2021-11-15:
Note doing this hack inside docker container works and this isn't required on the host, it works without it

```
cwd = os.getcwd()
os.chdir("/jetson-inference/data")
net = jetson.inference.poseNet("resnet18-body", None, threshold=0.15)
os.chdir(cwd)
```

dusty-nv, 2021-11-15:
Can you guys let me know which version of JetPack you are on?

I believe this issue is fixed in jetson-inference master branch, however I may have missed rebuilding the container for one of the JetPack versions that includes the fix.

zubairahmed-ai, 2021-11-15:
> Can you guys let me know which version of JetPack you are on?
> 
> I believe this issue is fixed in jetson-inference master branch, however I may have missed rebuilding the container for one of the JetPack versions that includes the fix.

@dusty-nv I am using `nvidia-jetpack_4.4` on host and the docker container I am using is `nvcr.io/nvidia/l4t-ml      r32.6.1-py3` due to opencv pre-installed on it

dusty-nv, 2021-11-15:
@zubairahmed-ai can you let me know what `cat /etc/nv_tegra_release` says?  Thanks.

zubairahmed-ai, 2021-11-15:
@dusty-nv It says 
```
nvidia@nvidia-desktop:~$ cat /etc/nv_tegra_release
# R32 (release), REVISION: 4.4, GCID: 23942405, BOARD: t210ref, EABI: aarch64, DATE: Fri Oct 16 19:44:43 UTC 2020
```
Could you please tell me all the commands you need to run in one shot, it seems our timezone difference isn't helping

dusty-nv, 2021-11-16:
Can you try pulling the latest and building it from source (outside container) like this:  https://github.com/dusty-nv/jetson-inference/blob/master/docs/building-repo-2.md

Then check to see if that fixes the issue when running it outside of container.  If it is, you can build the latest container with `docker/build.sh`

zubairahmed-ai, 2021-11-16:
@dusty-nv I have previously [struggled to build docker container from source before pulling the pre-built one](https://github.com/dusty-nv/jetson-containers/issues/101) I am not sure if this will end well this time around, do you think it will be any different?

zubairahmed-ai, 2021-11-16:
Note that outside of the container, posenet works fine and I am even able to use it, within the docker container [this hack above](https://github.com/dusty-nv/jetson-inference/issues/1295#issuecomment-968960829) fixes the loading issue but I am stuck with another issue during `process` method as it returns null within the docker container, I'll open a separate issue for that if required

dusty-nv, 2021-11-16:
OK, I just rebuilt the container for R32.4.4 - can you try a `sudo docker pull dustynv/jetson-inference:r32.4.4` ?

zubairahmed-ai, 2021-11-17:
Thanks @dusty-nv it works after pulling your docker build, I don't need to do [this hack](https://github.com/dusty-nv/jetson-inference/issues/1295#issuecomment-968960829) anymore

dusty-nv, 2021-11-17:
OK great @zubairahmed-ai.  I'm not sure if the container actually needed rebuilt or not (I kind of remember rebuilding it for R32.4.4 after I made that PoseNet fix) or if you just needed to do the `docker pull` to update your local copy of it.   Closing this issue now.

closed 2021-11-17 by dusty-nv