dusty-nv/jetson-inference #1296: Adding Function to detectNet.h Error

primetime728, 2021-11-12:
Hi,
I trained my own object detection and I am trying to interact with the detection. I changed the overlay to outline the object with a box and when the object is on the left/right of the screen the left/right side of the box is changed to red. 

```
/**
* Center Detection Flags
*/
inline void Allignment( int flag )					{ ALLIGN_FLAG = flag; }
```

It is declared to pass the flag from my python code into the detectNet.h.

```
# interact with detections on cam 0
for detection in detections_0:
	# print(detection)
	class_name = net.GetClassDesc(detection.ClassID)
	print(class_name + " Detected!")
	center = getCenter(detection)
	imgCenter = getImgCenter(display_0)
	coord_x = boxCoord(detection)

	# Right of Center
	if (coord_x > imgCenter[0] + 10):
		ALLIGNMENT = 0x1
		print("Right of Center")
	# Left of Center
	elif (coord_x < imgCenter[0] - 10):
		ALLIGNMENT = 0x2
		print("Left of Center")
	#Other
	else:
		ALLIGNMENT = 0x0
			
	print("-----------------------------------------")
	net.Allignment(ALLIGNMENT)
```

I declared and called 'Allignment' based on how the 'GetClassDesc' is documented. I use the 'make' and 'sudo make install' to compile the c++ and don't get any errors, but when I run my python program I get the error below.

Traceback (most recent call last):
  File "my_detection.py", line 141, in <module>
    main()
  File "my_detection.py", line 94, in main
    net.Allignment(ALLIGNMENT)
AttributeError: 'jetson.inference.detectNet' object has no attribute 'Allignment'

dusty-nv, 2021-11-15:
Hi @primetime728, did you add a Python binding for it in `jetson-inference/python/bindings/PyDetectNet.cpp` ?

https://github.com/dusty-nv/jetson-inference/blob/master/python/bindings/PyDetectNet.cpp

primetime728, 2021-11-18:
@dusty-nv thank you, it worked!

closed 2021-11-18 by primetime728