dusty-nv/jetson-inference #1299: built-in method Process of jetson.inference.poseNet object at 0x7fae89f510> returned NULL without setting an error

zubairahmed-ai, 2021-11-16:
Hi @dusty-nv 

When I am running the docker container and using `posenet` as below from the `/jetson-inference/data` folder I get the following error on `poses = net.Process(img, None, overlay='links,keypoints')` line please note that the `img` object is not none and this code works fine outside the container.

```
>>> poses = net.Process(img, overlay='links,keypoints')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
SystemError: <built-in method Process of jetson.inference.poseNet object at 0x7fae89f510> returned NULL without setting an error

```

Here's the complete code
```
import jetson.inference
import jetson.utils
import argparse
import time

parser = argparse.ArgumentParser(description="Run pose estimation DNN on a video/image stream.", formatter_class=argparse.RawTextHelpFormatter, epilog=jetson.inference.poseNet.Usage() + jetson.utils.videoSource.Usage() + jetson.utils.videoOutput.Usage() + jetson.utils.logUsage())

parser.add_argument("--overlay", type=str, default="links,keypoints", help="pose overlay flags (e.g. --overlay=links,keypoints)\nvalid combinations are:  'links', 'keypoints', 'boxes', 'none'")

net = jetson.inference.poseNet("resnet18-body", None, threshold=0.15)
img = cv2.imread("/jetson-inference/data/images/test.jpg")
poses = net.Process(img, overlay="links,keypoints")
cv2.imwrite('/pose_images/image-' + str(time.time()) + '.png', frame)
```

dusty-nv, 2021-11-16:
All of the DNN functions in jetson.inference expect a cudaImage (jetson.utils.cudaImage) and not an openCV image.

Try changing the code to this:

``` python
img = jetson.utils.loadImage("/jetson-inference/data/images/test.jpg")
poses = net.Process(img, overlay="links,keypoints")
jetson.utils.saveImage('/pose_images/image-' + str(time.time()) + '.png', frame)
```

zubairahmed-ai, 2021-11-17:
Thanks @dusty-nv this thought didn't cross my mind, it works but since I am not loading images from disk I did this `img = jetson.utils.cudaFromNumpy(img)` and `img = jetson.utils.cudaToNumpy(img)` instead

closed 2021-11-17 by zubairahmed-ai

dusty-nv, 2021-11-17:
OK cool, yes cudaFromNumpy() and cudaToNumpy() are also options for integrating it with OpenCV.