dusty-nv/jetson-inference #1300: Running "Hello AI world" on PC without Nvidia GPU TEGRA ?

fghoussen, 2021-11-17:
Is it possible to run the "Hello AI world" on PC _without_ Nvidia GPU TEGRA ?

I run a debian laptop
```
>> uname -a
Linux yoda 5.14.0-2-amd64 #1 SMP Debian 5.14.9-2 (2021-10-03) x86_64 GNU/Linux
>> sudo lshw -C display
  *-display                 
       description: VGA compatible controller
       product: 3rd Gen Core processor Graphics Controller
       vendor: Intel Corporation
       physical id: 2
       bus info: pci@0000:00:02.0
       logical name: /dev/fb0
       version: 09
       width: 64 bits
       clock: 33MHz
       capabilities: msi pm vga_controller bus_master cap_list fb
       configuration: depth=32 driver=i915 latency=0 resolution=1600,900
       resources: irq:30 memory:d0000000-d03fffff memory:c0000000-cfffffff ioport:2000(size=64) memory:c0000-dffff

``` 

This laptop has no NVidia GPU but runs an Intel Graphics shipped with the CPU.

I have a jetson-nano and flashed the latest jetpack (4.6 https://developer.nvidia.com/embedded/jetpack) on the micro SD card.

Running the very basic hello world (just running video-viewer) described here https://www.youtube.com/watch?v=QXIwdsyK7Rw&list=PL5B692fm6--uQRRDTPsJDp4o0xbzkoyf8&index=10 does not work as I first hit this error `head: cannot open '/etc/nv_tegra_release' for reading: no such file or directory`. To pass it I had to change docker scripts (which assume users have TEGRA GPU) this way (the error comes from `tools/l4t-version.sh` who checks for tegra driver version):
```

>> git diff
diff --git a/docker/tag.sh b/docker/tag.sh
index 456653b2..230a5740 100755
--- a/docker/tag.sh
+++ b/docker/tag.sh
@@ -1,7 +1,11 @@
 #!/usr/bin/env bash
 
 # find L4T_VERSION
-source tools/l4t-version.sh
+# source tools/l4t-version.sh
+L4T_VERSION=32.4.4
+L4T_RELEASE=32
+L4T_REVISION_MAJOR=4
+L4T_REVISION_MINOR=4
 
 # local container:tag name
 CONTAINER_IMAGE="jetson-inference:r$L4T_VERSION"
``` 

Then I had to install what I understand to be a piece of software needed to enable running NV GPU on docker container : 

 ```
>> distribution=debian10    && curl -s -L https://nvidia.github.io/nvidia-docker/gpgkey | sudo apt-key add -    && curl -s -L https://nvidia.github.io/nvidia-docker/$distribution/nvidia-docker.list | sudo tee /etc/apt/sources.list.d/nvidia-docker.list
>> sudo apt-get update && sudo apt-get install -y nvidia-docker2
>> sudo systemctl restart docker 
```

But this still fails :  

```
>> docker/run.sh
CONTAINER:     dustynv/jetson-inference:r32.4.4
DATA_VOLUME:   --volume /tmp/jetson-inference/data:/jetson-inference/data --volume /tmp/jetson-inference/python/training/classification/data:/jetson-inference/python/training/classification/data --volume /tmp/jetson-inference/python/training/classification/models:/jetson-inference/python/training/classification/models --volume /tmp/jetson-inference/python/training/detection/ssd/data:/jetson-inference/python/training/detection/ssd/data --volume /tmp/jetson-inference/python/training/detection/ssd/models:/jetson-inference/python/training/detection/ssd/models
USER_VOLUME:   
USER_COMMAND:  
V4L2_DEVICES:    --device /dev/video0  --device /dev/video1 
localuser:root being added to access control list
WARNING: The requested image's platform (linux/arm64) does not match the detected host platform (linux/amd64) and no specific platform was requested
docker: Error response from daemon: failed to create shim: OCI runtime create failed: container_linux.go:380: starting container process caused: process_linux.go:545: container init caused: Running hook #0:: error running hook: exit status 1, stdout: , stderr: nvidia-container-cli: initialization error: driver error: failed to process request: unknown.
>>

```

Is it possible to run the "Hello AI world" on PC _without_ Nvidia GPU TEGRA ?

Franck

dusty-nv, 2021-11-17:
Hi @fghoussen, it's not possible to run it without an NVIDIA GPU, because it uses CUDA, cuDNN, and TensorRT.

Further, I only build/test the project on Jetson.  There are others who have gotten it to build on x86, but they had an NVIDIA GPU in their system.

fghoussen, 2021-11-17:
OK, thanks to clarify this. I thought nvidia drivers (and likely CUDA) where shipped in the docker image, while jetpack was shipping software on jetson (SKD : cuDNN, TensorRT), so I supposed there was a way to get it to work. At least now I know there is no way.

Just to understand : CUDA, cuDNN and TensorRT must be installed on the host, are they ? But if so, what is the benefit of using docker ? What does jetpack can not ship that docker can ? Why not providing just a driver and a SDK (you could install on any distro) to use jetson-nano like you would have done for any other GPU or HW ? I don't get the logic here but I am sure this one!

dusty-nv, 2021-11-17:
On Jetson, currently CUDA/cuDNN/TensorRT are installed onto the Jetson, and docker mounts these in at runtime to keep the containers small.  Currently there are dependencies on the underlying kernel/GPU driver on Jetson, so this makes sense for Jetson for the most part.  However we are working on de-coupling these dependencies and moving towards the more traditional containerized model where CUDA/ect are installed inside the container.