dusty-nv/jetson-inference #1305: Nano 2GB stops training when retraining DetectNet image detection model

maxim739, 2021-11-24:
I am retraining the detectnet image detection model using my 2GB Jetson Nano, using:

"python3 train_ssd.py --dataset-type=voc --data=data/markers2 --model-dir=models/200markers2.5 --batch-size=2 --workers=1 --epochs=200"

But after about 9 errorless epochs into the training, the Jetson stops making progress (shown by no new epochs being trained, and Jetson resources being at idle). No error messages appear and the Jetson still has ample storage space. The jetson just seems to pause the training.

Has anyone else ran into this problem? I suspect it is a memory problem, but because I am new to linux I don't know what commands I should use to test these things.

dusty-nv, 2021-11-24:
Hi @maxim739, you may want to try decreasing the batch size to `--batch-size=1`

Also, have you disabled ZRAM, mounted SWAP, and disabled the desktop GUI?

* https://github.com/dusty-nv/jetson-inference/blob/master/docs/pytorch-transfer-learning.md#mounting-swap
* https://github.com/dusty-nv/jetson-inference/blob/master/docs/pytorch-transfer-learning.md#disabling-the-desktop-gui

maxim739, 2021-11-26:
I'll try that, thank you!

P.S Thanks for all the videos you make, they're really easy to understand and very helpful.

maxim739, 2021-11-26:
I'm also wondering if I could use the NVIDIA GPU on my windows computer to retrain an image detection model, convert it to onyx, and then run that model on my jetson. Is it the same architecture?

dusty-nv, 2021-11-29:
Hi @maxim739, I haven't tried PyTorch for Windows, but in theory yes you could.  I have run the training on my Ubuntu PC that has NVIDIA GPU (you would probably have better luck with Ubuntu).   You would need to install PyTorch/torchvision/ect on your PC.

You could clone this repo to your PC if you don't want to clone the whole jetson-inference:  https://github.com/dusty-nv/pytorch-classification
This repo is a submodule of jetson-inference

maxim739, 2021-11-30:
Thank you! I will look into that!

closed 2023-04-27 by dusty-nv