dusty-nv/jetson-inference #1311: Make docker containers persistent

dataplayer12, 2021-11-30:
It seems like containers launched by `docker/run.sh` script are automatically deleted when the user exits the container shell. Also, this means that any changes made within the container are never stored unless committed to an image. 

I would like to be able to see the container via `docker ps -a` and restart as I want to. How to do it?

zubairahmed-ai, 2021-12-02:
Same, except in my case I made a self-contained docker container that has this `jetson-inference` repo cloned within the container but every time I do `docker run` and loads it re-runs the pre-compilation below and its never persisted across `docker commit`

```
[TRT]    Fusing BatchNormalization_58 with Relu_59
[TRT]    Fusing BatchNormalization_47 with Relu_48
[TRT]    Fusing BatchNormalization_50 with Relu_51
[TRT]    Fusing BatchNormalization_61 with Relu_62
[TRT]    Fusing BatchNormalization_64 with Relu_65
[TRT]    Fusing BatchNormalization_53 with Relu_54
[TRT]    Fusing Sigmoid_56 with Mul_68
[TRT]    Fusing Tanh_67 with Mul_70
[TRT]    After vertical fusions: 39 layers
[TRT]    After final dead-layer removal: 39 layers
[TRT]    After tensor merging: 39 layers
[TRT]    After concat removal: 39 layers
[TRT]    Graph construction and optimization completed in 0.0155887 seconds
```

dataplayer12, 2021-12-02:
I solved the issue by removing --rm from the `run.sh` shell script. I launched the container with `run.sh` and made some changes. Terrified that these changes will be lost immediately upon exiting, I did a docker commit from outside the container and now I launch new containers from docker cli and don't use the shell scripts anymore. Hope this helps.

dataplayer12, 2021-12-02:
@zubairahmed-ai are you running docker within docker? That sounds like not a very good thing to do on a jetson.

dataplayer12, 2021-12-05:
Closing as `--rm` solves the issue.

closed 2021-12-05 by dataplayer12