dusty-nv/jetson-inference #1312: new problem about compiling on x86_64 PC

Kokoing123, 2021-12-04:
my env list follow this:
GTX16050TI
CUDA 10.2
TensorRT 7.1.3
Cudnn 8.0

The problem was:
[ 38%] Building CXX object utils/CMakeFiles/jetson-utils.dir/video/videoOptions.cpp.o
[ 38%] Building CXX object utils/CMakeFiles/jetson-utils.dir/video/videoOutput.cpp.o
[ 39%] Building CXX object utils/CMakeFiles/jetson-utils.dir/video/videoSource.cpp.o
[ 40%] Linking CXX shared library ../x86_64/lib/libjetson-utils.so
/usr/bin/ld: could not find -lnvbuf_utils
collect2: error: ld returned 1 exit status
utils/CMakeFiles/jetson-utils.dir/build.make:1134: recipe for target 'x86_64/lib/libjetson-utils.so' failed
make[2]: *** [x86_64/lib/libjetson-utils.so] Error 1
CMakeFiles/Makefile2:758: recipe for target 'utils/CMakeFiles/jetson-utils.dir/all' failed
make[1]: *** [utils/CMakeFiles/jetson-utils.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2

emmm, could you tell something about this problem?

dusty-nv, 2021-12-06:
Hi @Kokoing123, nvbuf_utils is a library that is only available for Jetson.   You would need to remove it from the various CMakeLists and compile with `cmake -DENABLE_NVMM=off`

Also, I only build this project for Jetson, I don't officially support x86 for it

RizM9292, 2022-03-25:
Hi, 
I am having the same issue so I tried this solution. However, I am getting the following error:

$ make -j$(nproc)
...........
CMakeFiles/Makefile2:367: recipe for target 'examples/detectnet/CMakeFiles/detectnet-camera.dir/all' failed
make[1]: *** [examples/detectnet/CMakeFiles/detectnet-camera.dir/all] Error 2
../../x86_64/lib/libjetson-utils.so: undefined reference to `NvBufferGetParams'
../../x86_64/lib/libjetson-utils.so: undefined reference to `NvDestroyEGLImage'
../../x86_64/lib/libjetson-utils.so: undefined reference to `ExtractFdFromNvBuffer'
../../x86_64/lib/libjetson-utils.so: undefined reference to `NvReleaseFd'
../../x86_64/lib/libjetson-utils.so: undefined reference to `NvEGLImageFromFd'
collect2: error: ld returned 1 exit status
examples/detectnet/CMakeFiles/detectnet.dir/build.make:101: recipe for target 'x86_64/bin/detectnet' failed
make[2]: *** [x86_64/bin/detectnet] Error 1
CMakeFiles/Makefile2:405: recipe for target 'examples/detectnet/CMakeFiles/detectnet.dir/all' failed
make[1]: *** [examples/detectnet/CMakeFiles/detectnet.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2

FYI, my env list is as follows:
GTX16050
CUDA 10.0
TensorRT 7.0

I know the package is built for the Jetson platform, but I need this repo to run another project (this is a dependency). I'd appreciate your response. Thank you!

dusty-nv, 2022-03-25:
Are you sure that you built with `cmake -DENABLE_NVMM=off` ?  Those functions shouldn't be used if that is turned off.

RizM9292, 2022-03-25:
Hi,
 Yes. I did the following. But it worked this time, after removing the old repo and restarting my system, and making a clean build. 
Here are the steps for reference:
-  I added my GPU to the root CMakeLists.txt architecture list,
-  I removed 'nvbuf_utils' from /utils/CMakeLists.txt
target_link_libraries(jetson-utils GL GLU GLEW gstreamer-1.0 gstapp-1.0 gstpbutils-1.0 nvbuf_utils ${CUDA_nppicc_LIBRARY})  to 
target_link_libraries(jetson-utils GL GLU GLEW gstreamer-1.0 gstapp-1.0 gstpbutils-1.0  ${CUDA_nppicc_LIBRARY}) 
- $ cmake -DENABLE_NVMM=off ../
- $ make -j$(nproc)

Thanks a lot for your prompt reply.