dusty-nv/jetson-inference #1313: SFTP to Jetson Nano

elongo, 2021-12-06:
Hi @dusty-nv , 
(I'll whine here, sorry, but...) is there a good workflow to upload files via SFTP to Jetson Nano. It has been a total nightmare that ubuntu distribution, it only add barriers and barriers to the workflow. There is no root user for ubuntu, and I can not find a way to download, edit, upload files via SFTP to the nano. I am coming from the Debian/Raspbian world which seems to be 1000 times more agile (?). Is this an accurate perception or just my ignorance? Have been working for 3 weeks on a project and still can't find a productive workflow with this machine, I only find "denied permission" everywhere. Kills creativity, not healthy :)). 

Any advice, guidance, links, would be immensely  appreciated.

Thank you,

elongo

dusty-nv, 2021-12-06:
Hi @elongo, since this topic is unrelated to this project, I would recommend posting your question to the Jetson Nano forum:

https://forums.developer.nvidia.com/c/agx-autonomous-machines/jetson-embedded-systems/jetson-nano/76

Personally, I use MobaXterm a lot for SSH/SFTP and it works without issue when the permissions are correct.  For editing root files I just use an editor over SSH with sudo.

elongo, 2021-12-06:
Thank you Dusty for the response,

Can you point me to a resource, tutorial or something to learn how the
proper permissions are set. I am supposed to be root, admin, everything,
but I only find restrictions to work with the nano.
I can normally work with all permission from Visual Studio Code, but that
has not been the case with Jetson Nano.

The only thing that I find is that with Ubuntu there isn't a root user,
although I can activate root from the command line, but that doesn"t seem
to help for the SFTP.

Thank you,

*Esteban Londoño G*
*Biodesign Lab** -School of Architecture + Design *
*Blacksburg, VA 24061*

closed 2023-04-27 by dusty-nv