dusty-nv/jetson-inference #1314: Cannot detect using SSD-mobilenet

qasmendoza, 2021-12-06:
I have concerns on my term project. I have 800+ data on this one and trained around 5 epochs (initial) but can’t seem to detect using the camera. Any idea why?
[image]

dusty-nv, 2021-12-06:
Hi @qasmendoza, can you check your PyTorch pth model (not the onnx) with this script:   https://github.com/dusty-nv/pytorch-ssd/blob/8ed842a408f8c4a8812f430cf8063e0b93a56803/run_ssd_example.py

If PyTorch model doesn't work, then it needs trained for longer.  I normally train detection models for at least 30 epochs.

Also, try deleting the .engine file in your model's folder whenever you re-export the ONNX.  Then TensorRT will generate the .engine again (otherwise it will be running off of old model)

qasmendoza, 2021-12-06:
@dusty-nv you are correct! The main issue is that the old .engine file is the one being loaded on the new workflow. Once I deleted it and re-export onnx after re-training the model for 50 epochs it worked! THANK YOU!

closed 2021-12-06 by qasmendoza