dusty-nv/jetson-inference #1315: getting nvbuffer as function

minusplusminus, 2021-12-06:
Hi,

I was wondering if it is possible to use a appsink signal seperately to get frame from gst? I know that `ExtractFdFromNvBuffer` is used to get buffer data.  how can I use this lib to request the buffer instead of `sample.get_buffer()`? 

python
```
sample = sink.emit("pull-sample") 
if isinstance(sample, Gst.Sample):

        buffer = sample.get_buffer()  # Gst.Buffer

        caps_format = sample.get_caps().get_structure(0)  # Gst.Structure

        # GstVideo.VideoFormat

        video_format = GstVideo.VideoFormat.from_string(
            caps_format.get_value('format'))

        w, h = caps_format.get_value('width'), caps_format.get_value('height')
        c = utils.get_num_channels(video_format)

        buffer_size = buffer.get_size()
        format_info = GstVideo.VideoFormat.get_info(video_format)  # GstVideo.VideoFormatInfo
        array = np.ndarray(shape=buffer_size // (format_info.bits // utils.BITS_PER_BYTE),
                           buffer=buffer.extract_dup(0, buffer_size),
                           dtype=utils.get_np_dtype(video_format))
```

dusty-nv, 2021-12-06:
Hi @minusplusminus, I'm not sure how to use nvbuf_utils library from Python, because it's a C library.

minusplusminus, 2021-12-06:
https://github.com/dusty-nv/jetson-utils/blob/0d3f59f5c0967a108ec1cd58a518ee5ad817d35b/codec/gstBufferManager.cpp#L159 
shows that it's used, so I thought it could maybe be accessible through a custom pything binding

dusty-nv, 2021-12-06:
If you wanted to write custom python bindings for it, then sure that could be an option.  Such bindings do not exist for it already.  If you do it, I would be careful about how the memory is handled in Python after you return it, because it could negate the positives of using NVMM in the first place (zero-copy)

minusplusminus, 2021-12-08:
yes, i'm familiar with the logic behind it. I did some research on (R)DMA buffer in the past. Sadly enough NVMM is not supported by the closed source lib i'm using :(

closed 2023-04-27 by dusty-nv