dusty-nv/jetson-inference #1316: -bash: video-viewer: command not found

fghoussen, 2021-12-07:
Seems no `video-viewer` on nano when ssh'ing from pc ?

```
me@pc:~$ ssh me@192.168.0.xx (this is nano IP)
Welcome to Ubuntu 18.04.5 LTS (GNU/Linux 4.9.253-tegra aarch64)

 * Documentation:  https://help.ubuntu.com
 * Management:     https://landscape.canonical.com
 * Support:        https://ubuntu.com/advantage
This system has been minimized by removing packages and content that are
not required on a system that users do not log into.

To restore this content, you can run the 'unminimize' command.

160 updates can be applied immediately.
108 of these updates are standard security updates.
To see these additional updates run: apt list --upgradable

Last login: Tue Dec  7 07:55:19 2021 from 192.168.0.13
-bash: /usr/bin/tclsh: No such file or directory
-bash: export: _moduleraw: not a function
-bash: export: module: not a function
me@nano:~$ video-viewer
-bash: video-viewer: command not found

```

What did I miss ?

dusty-nv, 2021-12-07:
When you built jetson-inference, did you do `sudo make install`?

Is there video-viewer found under `/usr/local/bin`?

fghoussen, 2021-12-08:
Oh men! Didn't get there was some stuff to compile: is `mkdir build; cd build; cmake ..; sudo make all install` enough to get `video-viewer`, `detectnet` or others? I followed the "hello IA world" youtube video which uses docker (KO as I don't have NV GPU on my laptop) so I had to ssh directly and didn't find anything there (I thought it was part of jetpack, but I guess the `docker/run.sh` script build all this from jetson-inference repo: correct!?...). On the road now, I'll be able to test only tomorrow: would be great if you could detail how to do (cmake options, ...) to get binaries and code (python, c++) ready to use!

dusty-nv, 2021-12-08:
Sorry, but this project only officially supports Jetson/ARM platform, as there are some Jetson-specific features and libraries used.  Some other users have figured it out for x86, but if you don't have an NVIDIA GPU it wouldn't work anyways as this project uses CUDA and other NVIDIA libraries like TensorRT.

fghoussen, 2021-12-09:
OK, so as I do not have NV GPU on my PC, even ssh'ing from my PC to the nano will not work, right?

dusty-nv, 2021-12-09:
If you have a Nano, then yep SSH from your PC into the Nano will work because its actually running on the Nano - except for the OpenGL display.

dusty-nv, 2021-12-10:
Can you post the rest of the terminal log from when you run video-viewer?   The problem doesn't seem related to OpenGL, it seems related to your camera.  What camera are you using?

fghoussen, 2021-12-12:
Not sure what type of camera I have, but, I would say a USB one (not sure this helps anyway).

With my USB camera, I get 

```
me@nano:~$ v4l2-ctl --list-devices
USB 2.0 Camera (usb-70090000.xusb-2.2):
	/dev/video0

me@nano:~$ v4l2-ctl --list-formats-ext
ioctl: VIDIOC_ENUM_FMT
	Index       : 0
	Type        : Video Capture
	Pixel Format: 'MJPG' (compressed)
	Name        : Motion-JPEG
		Size: Discrete 1280x720
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 640x480
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 352x288
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 320x240
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 176x144
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 160x120
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 800x600
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 960x720
			Interval: Discrete 0.033s (30.000 fps)

	Index       : 1
	Type        : Video Capture
	Pixel Format: 'YUYV'
	Name        : YUYV 4:2:2
		Size: Discrete 1280x720
			Interval: Discrete 0.100s (10.000 fps)
		Size: Discrete 640x480
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 352x288
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 320x240
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 176x144
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 160x120
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 800x600
			Interval: Discrete 0.050s (20.000 fps)
		Size: Discrete 960x720
			Interval: Discrete 0.067s (15.000 fps)

me@nano:~/Workspaces/jetson-inference/build$ video-viewer /dev/video0 rtp://192.168.0.13:1234 2>&1 usb.log
```
[usb.log](https://github.com/dusty-nv/jetson-inference/files/7699166/usb.log)

dusty-nv, 2021-12-13:
Can you try running with the `--headless` flag like so?

```
video-viewer --headless /dev/video0 rtp://192.168.0.13:1234
```

Also the `gstreamer message qos ==> jpegdec0` message typically indicate some problem with the USB connection or bandwidth.  Does the issue persist if you run it at a lower resolution?

```
video-viewer --headless --input-width=320 --input-height=240 /dev/video0 rtp://192.168.0.13:1234
```

You can also try disabling MJPEG with `--input-codec=raw`

fghoussen, 2021-12-13:
Works at nano side (no error), but, still freeze at PC side (hang, no video pop-up) :

```
>> gst-launch-1.0 -v udpsrc port=1234  caps = "application/x-rtp, media=(string)video, clock-rate=(int)90000, encoding-name=(string)H264, payload=(int)96" !  rtph264depay ! decodebin ! videoconvert ! autovideosink
Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
/GstPipeline:pipeline0/GstUDPSrc:udpsrc0.GstPad:src: caps = application/x-rtp, media=(string)video, clock-rate=(int)90000, encoding-name=(string)H264, payload=(int)96
New clock: GstSystemClock
/GstPipeline:pipeline0/GstRtpH264Depay:rtph264depay0.GstPad:sink: caps = application/x-rtp, media=(string)video, clock-rate=(int)90000, encoding-name=(string)H264, payload=(int)96

```

dusty-nv, 2021-12-13:
Is the IP address of your PC indeed `192.168.0.13` ?

Can you check to make sure that the received packets on your PC are increasing?   Is there a firewall blocking it perhaps?   You may want to try another port like 5000

fghoussen, 2021-12-13:
Weird. The IP of my PC is indeed `192.168.0.13` over wifi on my local network. The nano is on LAN (eth0 `192.168.0.12`). Using port 5000 does not fix. No idea why (I expected this to work)...

dusty-nv, 2021-12-13:
You can try running this from the Jetson as a test pattern:

``` bash
gst-launch-1.0 -v videotestsrc ! video/x-raw,width=300,height=300,framerate=30/1 ! x264enc ! rtph264pay ! udpsink host=192.168.0.13 port=5000
```

fghoussen, 2021-12-15:
Turns out FPS drops to 1-3 when ssh'ing, but, when displaying directly on nano I get FPS about 15+ ?!... Didn't expected that

dusty-nv, 2021-12-15:
Are you trying to display the video over SSH / X11 tunneling?   That won't give good performance for full-motion video.  When displaying directly on Nano, it uses OpenGL.   

RTP is the preferred way to display the video remotely if you can get it working.  Were you able to stream the test pattern from my last comment from your Nano to your PC?   If not, my suspicion is there is some networking issue occurring.

fghoussen, 2021-12-17:
Missed your reply: mail overflow!

> Are you trying to display the video over SSH / X11 tunneling?

Would say yes, always connecting with `ssh -X` as otherwise you don't get display. So If as I suspect X means X11, I guess yes

> Were you able to stream the test pattern

Frezzing.... Nothing pops up (no video window)

On nano sshing without -X

```
me@nano:~$ gst-launch-1.0 -v videotestsrc ! video/x-raw,width=300,height=300,framerate=30/1 ! x264enc ! rtph264pay ! udpsink host=192.168.0.13 port=5000
Setting pipeline to PAUSED ...
Pipeline is PREROLLING ...
/GstPipeline:pipeline0/GstVideoTestSrc:videotestsrc0.GstPad:src: caps = video/x-raw, format=(string)I420, width=(int)300, height=(int)300, framerate=(fraction)30/1, multiview-mode=(string)mono, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive
/GstPipeline:pipeline0/GstCapsFilter:capsfilter0.GstPad:src: caps = video/x-raw, format=(string)I420, width=(int)300, height=(int)300, framerate=(fraction)30/1, multiview-mode=(string)mono, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive
Redistribute latency...
/GstPipeline:pipeline0/GstX264Enc:x264enc0.GstPad:sink: caps = video/x-raw, format=(string)I420, width=(int)300, height=(int)300, framerate=(fraction)30/1, multiview-mode=(string)mono, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive
/GstPipeline:pipeline0/GstCapsFilter:capsfilter0.GstPad:sink: caps = video/x-raw, format=(string)I420, width=(int)300, height=(int)300, framerate=(fraction)30/1, multiview-mode=(string)mono, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive
/GstPipeline:pipeline0/GstX264Enc:x264enc0.GstPad:src: caps = video/x-h264, codec_data=(buffer)01640014ffe1001d67640014acd941309fbbc05a830082d280000003008000001e478a14cb01000568ebecb22c, stream-format=(string)avc, alignment=(string)au, level=(string)2, profile=(string)high, width=(int)300, height=(int)300, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)30/1, interlace-mode=(string)progressive, colorimetry=(string)bt601, chroma-site=(string)jpeg, multiview-mode=(string)mono, multiview-flags=(GstVideoMultiviewFlagsSet)0:ffffffff:/right-view-first/left-flipped/left-flopped/right-flipped/right-flopped/half-aspect/mixed-mono
/GstPipeline:pipeline0/GstRtpH264Pay:rtph264pay0.GstPad:src: caps = application/x-rtp, media=(string)video, clock-rate=(int)90000, encoding-name=(string)H264, packetization-mode=(string)1, profile-level-id=(string)640014, sprop-parameter-sets=(string)"Z2QAFKzZQTCfu8BagwCC0oAAAAMAgAAAHkeKFMs\=\,aOvssiw\=", payload=(int)96, ssrc=(uint)595729847, timestamp-offset=(uint)3413087185, seqnum-offset=(uint)23944, a-framerate=(string)30
/GstPipeline:pipeline0/GstUDPSink:udpsink0.GstPad:sink: caps = application/x-rtp, media=(string)video, clock-rate=(int)90000, encoding-name=(string)H264, packetization-mode=(string)1, profile-level-id=(string)640014, sprop-parameter-sets=(string)"Z2QAFKzZQTCfu8BagwCC0oAAAAMAgAAAHkeKFMs\=\,aOvssiw\=", payload=(int)96, ssrc=(uint)595729847, timestamp-offset=(uint)3413087185, seqnum-offset=(uint)23944, a-framerate=(string)30
/GstPipeline:pipeline0/GstRtpH264Pay:rtph264pay0.GstPad:sink: caps = video/x-h264, codec_data=(buffer)01640014ffe1001d67640014acd941309fbbc05a830082d280000003008000001e478a14cb01000568ebecb22c, stream-format=(string)avc, alignment=(string)au, level=(string)2, profile=(string)high, width=(int)300, height=(int)300, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)30/1, interlace-mode=(string)progressive, colorimetry=(string)bt601, chroma-site=(string)jpeg, multiview-mode=(string)mono, multiview-flags=(GstVideoMultiviewFlagsSet)0:ffffffff:/right-view-first/left-flipped/left-flopped/right-flipped/right-flopped/half-aspect/mixed-mono
/GstPipeline:pipeline0/GstRtpH264Pay:rtph264pay0: timestamp = 3413087185
/GstPipeline:pipeline0/GstRtpH264Pay:rtph264pay0: seqnum = 23944
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock

```

And on my PC

```
>> gst-launch-1.0 -v udpsrc port=5000  caps = "application/x-rtp, media=(string)video, clock-rate=(int)90000, encoding-name=(string)H264, payload=(int)96" !  rtph264depay ! decodebin ! videoconvert ! autovideosink
Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
/GstPipeline:pipeline0/GstUDPSrc:udpsrc0.GstPad:src: caps = application/x-rtp, media=(string)video, clock-rate=(int)90000, encoding-name=(string)H264, payload=(int)96
/GstPipeline:pipeline0/GstRtpH264Depay:rtph264depay0.GstPad:sink: caps = application/x-rtp, media=(string)video, clock-rate=(int)90000, encoding-name=(string)H264, payload=(int)96
New clock: GstSystemClock

```

Exact same happens if I use `ssh -X`

fghoussen, 2021-12-17:
On my PC, does iptable config could block the video stream?

```
>> more /etc/network/iptables.up.rules
# Generated by xtables-save v1.8.1 on Sat Nov  3 19:02:11 2018
*filter
:INPUT DROP [0:0]
:FORWARD DROP [0:0]
:OUTPUT ACCEPT [0:0]
-A INPUT -i lo -j ACCEPT
-A INPUT -p icmp -j DROP
-A INPUT -p udp -m udp --sport 53 -m conntrack --ctstate RELATED,ESTABLISHED -j ACCEPT
-A INPUT -p tcp -m multiport --sports 80,443,8000 -m conntrack --ctstate RELATED,ESTABLISHED -j ACCEPT
-A INPUT -p tcp -m multiport --sports 25,2525,587,465,143,993,995 -m conntrack --ctstate RELATED,ESTABLISHED -j ACCEPT
-A INPUT -p tcp -m tcp --dport 22 -m conntrack --ctstate RELATED,ESTABLISHED -j ACCEPT
-A INPUT -m conntrack --ctstate RELATED,ESTABLISHED -j ACCEPT
-A INPUT -j LOG --log-prefix "INPUT:   "
-A FORWARD -j LOG --log-prefix "FORWARD: "
COMMIT
# Completed on Sat Nov  3 19:02:11 2018
# Generated by xtables-save v1.8.1 on Sat Nov  3 19:02:11 2018
*nat
:PREROUTING ACCEPT [0:0]
:INPUT ACCEPT [0:0]
:POSTROUTING ACCEPT [0:0]
:OUTPUT ACCEPT [0:0]
COMMIT
# Completed on Sat Nov  3 19:02:11 2018
# Generated by xtables-save v1.8.1 on Sat Nov  3 19:02:11 2018
*mangle
:PREROUTING ACCEPT [0:0]
:INPUT ACCEPT [0:0]
:FORWARD ACCEPT [0:0]
:OUTPUT ACCEPT [0:0]
:POSTROUTING ACCEPT [0:0]
COMMIT
# Completed on Sat Nov  3 19:02:11 2018

```

dusty-nv, 2021-12-17:
> Would say yes, always connecting with `ssh -X` as otherwise you don't get display. So If as I suspect X means X11, I guess yes

The display from jetson-inference won't work even with `ssh -X`, because it is OpenGL rendering.   Your issue appears to be unrelated to X11 tunneling anyways, since the test pipeline you ran is still showing on the PC.   I'm not sure what the networking issue is as that's not my area of the expertise.

fghoussen, 2021-12-17:
> that's not my area of the expertise.

Neither mine!... Thanks to have taken time to look at that.

closed 2021-12-17 by fghoussen

emma-jinger, 2022-07-05:
> > that's not my area of the expertise.
> 
> Neither mine!... Thanks to have taken time to look at that.

Hi fghoussen, I have run into the exact issue as you did ([gstreamer] gstEncoder::Render() -- unsupported image format (rgb8)).
Have you figured out how to solve the problem?

fghoussen, 2022-07-07:
@emma-jinger: try to flush iptables.