dusty-nv/jetson-inference #1317: Issues about FPS during multi tasks

zhuzhuxia1221, 2021-12-07:
I ran two tasks (both cityscapes-512x256) simultaneously on Jetson Xavier and then turned out that FPS doubled compared to running a single task. However, Then I increased the resolution to 2048x1024 and it resulted in FPS cutting down a half when I launched another task. I wonder why it happened. The screenshots are attached to this comment.

[image]

[image]

[image]

[image]

Looking forward to u reply.

dusty-nv, 2021-12-07:
I think it's because the more complex 2048x1024 model uses 100% of GPU, and hence it reduces framerate when you run multiple.

You can keep an eye on the GPU utilization using the `tegrastats` utility.

zhuzhuxia1221, 2021-12-08:
Thank you for your reply.
Did you mean that the function of net.GetNetworkFPS() actually means the sum of the frames of all the running tasks?

dusty-nv, 2021-12-08:
`net.GetNetworkFPS()` returns how long it took to process that particular model with TensorRT.   However if you have other processes running using the GPU, the GPU may do time-sharing between them, so it can slow down a model.   

If you wrote a single application/script that ran the same model twice on different streams, then I believe you should see normal performance reported from `net.GetNetworkFPS()` because the models would be executed in serial as opposed to in parallel.

closed 2023-04-27 by dusty-nv