dusty-nv/jetson-inference #1319: Re-trained SSD-Mobilenet does not detect fruit

DariusWigglesworth, 2021-12-09:
I have been following the steps for re-training SSD-Mobilenet on the fruit dataset from the github page [here](https://github.com/dusty-nv/jetson-inference/blob/master/docs/pytorch-ssd.md) and tutorial video [here](https://www.youtube.com/watch?v=QXIwdsyK7Rw&list=PL5B692fm6--uQRRDTPsJDp4o0xbzkoyf8&index=9) on my `dustynv/jetson-inference:r32.5.0` docker container on a 4gb Jetson Nano. However, after I've re-trained the model on the fruit set and run it on the fruit pictures in `/jetson-inference/data/images` none of the pictures have the bounding box, label, or confidence number. 

I noticed in the tutorial video that Dusty's terminal looks like this when he runs his model after training it on the fruit:
[image]
However, my terminal does not have the lines in the red `[`, only `[image] loaded /jetson-inference/data/images/fruit_i.jpg` and `[image] saved /jetson-inference/data/images/test/fruit_i.jpg`. Therefore, it appears to me that it isn't detecting any fruit at all.

I have tried re-training the model, re-downloading the fruit dataset and re-training the model on that, training the model on 5 epochs instead of 1, and cloning a new container and training the model on the fruit data works. However, nothing has fixed this issue of the fruit not being detected. I copied the commands from the github page so it is unlikely that the command is wrong. I didn't have an issue with the bounding boxes on the previous tutorial with the cats & dogs and objects. What else can I try to fix this issue?

dusty-nv, 2021-12-09:
Hi @DariusWigglesworth, can you try deleting the cached *.engine file from your model's folder?

Also, can you try my fruit model from here:  https://nvidia.box.com/shared/static/gq0zlf0g2r258g3ldabl9o7vch18cxmi.gz
It was trained for 100 epochs, although 30 should be sufficient I believe.

DariusWigglesworth, 2021-12-13:
Hi @dusty-nv thanks for your assistance. I tried deleting the *.engine file from the model folder and I got the same result with the fruit without a bounding box. However, when I run your 100 epoch model I get the bounding boxes present on my fruit. Could this mean that there is an issue with how my model is being trained on the fruit dataset?

dusty-nv, 2021-12-13:
I think that your model needed trained for more epochs.  If you do that, remember to delete the *.engine file again after re-exporting your ONNX model.   Or you may choose to move on to training a model on your own custom dataset.

DariusWigglesworth, 2021-12-15:
Thanks for your help Dusty, yesterday I trained it for 30 epochs and after I ran it I was able to see the bounding boxes and confidence scores on the images. Did I do anything wrong with the training to cause it to differ from your example? Previously when following the examples my output matched yours exactly. How come you were able to train it for one epoch and get an output when I had to train it for more?

dusty-nv, 2021-12-15:
Hi Darius, I only trained it for 1 epoch in the video to speed up the video, I actually ran the model that was trained for more epochs.  I believe I (hopefully) mentioned this in the video.

DariusWigglesworth, 2021-12-16:
Oh I see, thank you for clarifying. And thank you for all you help with this issues as well as creating the awesome tutorials!

closed 2021-12-16 by DariusWigglesworth