dusty-nv/jetson-inference #1321: Network training made AGX Xavier crash, now gves EOFError: Ran out of input

jedynygucio, 2021-12-10:
Good afternoon,

I was following the retraining example on some of my own data and when the python3 train.py (with defaults) etc. command was executed my Jetson completely crashed.

Upon reboot, it appears to have somehow 'corrupted' the resnet18 model it was working on, because now it gives this message, even on the cat-dog dataset:

Use GPU: 0 for training
=> dataset classes:  2 ['cat', 'dog']
=> using pre-trained model 'resnet18'
Traceback (most recent call last):
  File "train.py", line 506, in <module>
    main()
  File "train.py", line 135, in main
    main_worker(args.gpu, ngpus_per_node, args)
  File "train.py", line 199, in main_worker
    model = models.__dict__[args.arch](pretrained=True)
  File "/usr/local/lib/python3.6/dist-packages/torchvision-0.7.0a0+78ed10c-py3.6-linux-aarch64.egg/torchvision/models/resnet.py", line 241, in resnet18
    **kwargs)
  File "/usr/local/lib/python3.6/dist-packages/torchvision-0.7.0a0+78ed10c-py3.6-linux-aarch64.egg/torchvision/models/resnet.py", line 227, in _resnet
    progress=progress)
  File "/home/jedynygucio/.local/lib/python3.6/site-packages/torch/hub.py", line 495, in load_state_dict_from_url
    return torch.load(cached_file, map_location=map_location)
  File "/home/jedynygucio/.local/lib/python3.6/site-packages/torch/serialization.py", line 585, in load
    return _legacy_load(opened_file, map_location, pickle_module, **pickle_load_args)
  File "/home/jedynygucio/.local/lib/python3.6/site-packages/torch/serialization.py", line 755, in _legacy_load
    magic_number = pickle_module.load(f, **pickle_load_args)
EOFError: Ran out of input

Running with another model e.g. googlenet resulted with another crash, and then this model because 'corrupted' an no longer shows the above. I'm running on the AGX Xavier so there shouldn't be any issues with hardware being overburdened?

Any advice would be greatly appreciated.

jedynygucio, 2021-12-10:
To add, I tried reinstalling pytorch and it did not fix this. 
torch.__version__ gives 1.6.0
torchvision.__version__ gives 0.7.0a0+78ed10c

dusty-nv, 2021-12-10:
Hi @jedynygucio, are you using the Jetson AGX Xavier devkit?    Do you have enough free disk space on the system?

I believe the PyTorch/torchvision base models get cached to `~/.cache/torch` if you want to try deleting that directory, so they get re-downloaded

https://discuss.pytorch.org/t/how-to-delete-loaded-model-from-torch-hub/74445/2

jedynygucio, 2021-12-10:
Hello @dusty-nv thank you very much for your very prompt reply. I am using the AGX Xavier devkit yes. The link you suggested did indeed help to clear the cache and I could run the cat-dog example successfully for the first time. 

I am now transfer training resnet18 on a dataset of mine of some 4000 .png images, I'm on the 3rd epoch and it sems to now be running fine. I don't know what the crashing issue was. It's drawing up to 38 W on MAXN and hitting 62 degrees C.

Is there possibly a way to train a model from scratch, just to see it improve?

dusty-nv, 2021-12-10:
You can try changing this line of code from train.py to `default=False`  

https://github.com/dusty-nv/pytorch-classification/blob/393914b484fc946876e0a1fe37c971acdfb1fec6/train.py#L76

jedynygucio, 2021-12-10:
By the way, in your github pages you mention to use the `--epoch-start` flag to resume training but I think it should be `--start-epoch` :)
https://github.com/dusty-nv/jetson-inference/blob/master/docs/pytorch-collect.md

jedynygucio, 2021-12-10:
Thank you for the info on the untrained model, I would like to give it a shot.

jedynygucio, 2021-12-10:
@dusty-nv  changing line 76 as suggested doesn't seem to have done anything different - it's still loading is resnet18 (with all the input / hidden / output layers etc.) so how can it not be pre-trained anyway?

closed 2023-04-27 by dusty-nv