dusty-nv/jetson-inference #1322: Including jetson.utils in a systemd service file results in core-dump

TimothyBaalman, 2021-12-12:
I can execute my python script and have it run perfectly with 
`sudo python3 <filename>.py`

but with a service file that is setup like
```vim
[Unit]
Description=Python Boot Service File

[Service]
WorkingDirectory=/home/<project root>
ExecStart=/usr/bin/python3 /home/<project root>/main.py

[Install]
WantedBy=multi-user.target
```
results in the following errors
```
Main process exited, code=dumped, status=4/ILL
Failed with result 'core-dump'
```

I also know that the problem is including jetson.utils since a made a test.py file that will just do and print some math operations. It worked using the same service file setup until I included jetson.utils. 

Is there some special pathing that I am missing to be able to use jetson.utils and jetson.inference in a python file that is setup as a service file?

luozy-usc, 2022-03-31:
Have you find the solution? I guess I've met the same problem.

luozy-usc, 2022-03-31:
> I can execute my python script and have it run perfectly with `sudo python3 <filename>.py`
> 
> but with a service file that is setup like
> 
> ```viml
> [Unit]
> Description=Python Boot Service File
> 
> [Service]
> WorkingDirectory=/home/<project root>
> ExecStart=/usr/bin/python3 /home/<project root>/main.py
> 
> [Install]
> WantedBy=multi-user.target
> ```
> 
> results in the following errors
> 
> ```
> Main process exited, code=dumped, status=4/ILL
> Failed with result 'core-dump'
> ```
> 
> I also know that the problem is including jetson.utils since a made a test.py file that will just do and print some math operations. It worked using the same service file setup until I included jetson.utils.
> 
> Is there some special pathing that I am missing to be able to use jetson.utils and jetson.inference in a python file that is setup as a service file?

Have you found the solution? I guess I've met the same problem.

BeMain, 2023-08-30:
Why is this closed? I encountered the same problem today. Did anyone find a solution?

TimothyBaalman, 2023-08-30:
I did not find a solution to this problem. This was a school project that I was working that it wasn’t a requirement to get working. It would have been nice to get working. Not sure why it’s closed though.