dusty-nv/jetson-inference #1323: How to get latest RTSP frame using gstreamer (without OpenCV)?

leenremm, 2021-12-12:
Hi @dusty-nv 

I am connecting my Jetson Nano to a RTSP stream (Dahua camera).

The Jetson Nano is processing frames at roughly 10 FPS (9.8 - 10.2 FPS). The IP camera RTSP stream is fixed at 30 FPS. Currently the gstreamer query gets the "next" frame in the sequence, meaning that the Jetson Nano is always processing old images. For example, after 10 seconds the IP Camera RTSP stream has 300 images in the buffer, and the Jetson Nano is still processing image #100. 

With OpenCV there are ways around this. For example: # https://stackoverflow.com/questions/43665208/how-to-get-the-latest-frame-from-capture-device-camera-in-opencv

But we avoid using OpenCV with the jetson-inference repo, for performance reasons.

Any suggestions for how I can configure gstreamer to get the latest image from the RTSP stream?

dusty-nv, 2021-12-13:
Hi @leenremm, the videoSource interface always returns to you the latest frame that it got from GStreamer, so the latency is coming from inside the GStreamer pipeline.  Can you try adding [`drop-on-latency=true`](https://groups.google.com/g/kurento/c/ptWOqxnzc9Q/m/bFKVQl7wCQAJ) to the `rtspsrc` element?

https://github.com/dusty-nv/jetson-utils/blob/0d3f59f5c0967a108ec1cd58a518ee5ad817d35b/codec/gstDecoder.cpp#L527

Try changing this to:  `ss << " latency=" << mOptions.rtspLatency << " drop-on-latency=true";`

Then run:

``` bash
cd /path/to/your/jetson-inference/build
cmake ../
make
sudo make install
```

If that helps fix it, please let me know.

AK51, 2021-12-16:
Hi,

I have tried it, but the delay is still there....
video-viewer --input-rtsp-latency=0 --input-codec=h264 rtsp//...........

closed 2023-04-27 by dusty-nv