dusty-nv/jetson-inference #1327: How to get detectNet() to accept bgr8 format images?

leenremm, 2021-12-16:
I have a model trained with BGR images (separate to the Jetson code), and jetson inference detectNet() accepts RGB images. I need to convert the RGB images taken by the Capture / image load script into BGR format before passing them to the 

When I try to convert the colour.

img = input.Capture()
bgr_img = jetson.utils.cudaAllocMapped(width=img.width, height=img.height, format='bgr8')
jetson.utils.cudaConvertColor(rgb_img, bgr_img)
objects_detected = net.Detect(bgr_img)

This correctly switched the R and the B:

```
(Pdb) bgr_img[0]
(52, 66, 83)
(Pdb) rgb_img[0]
(83, 66, 52)
```

But detectNet does not allow bgr8 format.

```
[TRT]    detectNet::Detect() -- unsupported image format (bgr8)
[TRT]                           supported formats are:
[TRT]                              * rgb8
[TRT]                              * rgba8
[TRT]                              * rgb32f
[TRT]                              * rgba32f
*** Exception: jetson.inference -- detectNet.Detect() encountered an error classifying the image
```

The work-around is to export / import from Numpy:

img = input.Capture()
img_np = jetson.utils.cudaToNumpy(img, img.width, img.height, img.channels)
img_np_new = img_np[...,::-1].copy()
new_img = jetson.utils.cudaFromNumpy(img_np_new)
objects_detected = net.Detect(new_img, overlay=overlay)

But this takes long (an additional 62ms per image)!

How can I trick the detectNet.Detect() to allow me to pass bgr8 format (which is really rgb8)?

leenremm, 2021-12-16:
@dusty-nv your feedback would be greatly appreciated!

leenremm, 2021-12-16:
Adding pixel by pixel manipulation takes too long (> 10 seconds)

```
def convert_rgb_to_bgr(cuda_img):
    for y in range(cuda_img.height):
        for x in range(cuda_img.width):
            cuda_img[y,x] = cuda_img[y,x][::-1] # RGB <=> BGR conversion
    return cuda_img
```

dusty-nv, 2021-12-16:
@leenremm you can try changing the lower-level source to detectNet.cpp here:

https://github.com/dusty-nv/jetson-inference/blob/9b66fbf38dcd0e81f37eb3748db3df0a221b4fa7/c/detectNet.cpp#L689

You would probably need to do this anyways to update the pre/post-processing to reflect what your model expects.

Also, if you change the cpp source code, run this to rebuild/reinstall:

``` bash
cd /path/to/jetson-inference/build
cmake ../
make
sudo make install
```

leenremm, 2021-12-16:
Thanks @dusty-nv. I'll give that a go.

leenremm, 2021-12-16:
That is an elegant fix, thanks @dusty-nv 

Final solution was to add:

```
  if (format == IMAGE_BGR8)
  {
    format = IMAGE_RGB8;
  }

```

Just before:

```
	if( !imageFormatIsRGB(format) )
	{
		LogError(LOG_TRT "detectNet::Detect() -- unsupported image format (%s)\n", imageFormatToStr(format));
		LogError(LOG_TRT "                       supported formats are:\n");
		LogError(LOG_TRT "                          * rgb8\n");
		LogError(LOG_TRT "                          * rgba8\n");
		LogError(LOG_TRT "                          * rgb32f\n");
		LogError(LOG_TRT "                          * rgba32f\n");

		return -1;
	}
```

In detectNet.cpp

leenremm, 2021-12-16:
Note: It's a hack for the specific use case I was facing. Not a suggested fix / feature.

closed 2021-12-16 by leenremm