dusty-nv/jetson-inference #1328: Traceback error

enigma460, 2021-12-17:
Hi, 
When I run the Training Image Classification Models video example, I get a traceback error.
This is at the end of the Epoch computation:

Epoch: [0][1240/1250]  Time  0.363 ( 0.387)  Data  0.001 ( 0.021)  Loss 7.1227e-01 (1.3036e+00)  Acc@1  50.00 ( 51.09)  Acc@5 100.00 (100.00)
Epoch: [0] completed, elapsed time 483.332 seconds
Traceback (most recent call last):
  File "train.py", line 506, in <module>
    main()
  File "train.py", line 135, in main
    main_worker(args.gpu, ngpus_per_node, args)
  File "train.py", line 280, in main_worker
    acc1 = validate(val_loader, model, criterion, num_classes, args)
  File "train.py", line 372, in validate
    for i, (images, target) in enumerate(val_loader):
  File "/usr/local/lib/python3.6/dist-packages/torch/utils/data/dataloader.py", line 359, in __iter__
    return self._get_iterator()
  File "/usr/local/lib/python3.6/dist-packages/torch/utils/data/dataloader.py", line 305, in _get_iterator
    return _MultiProcessingDataLoaderIter(self)
  File "/usr/local/lib/python3.6/dist-packages/torch/utils/data/dataloader.py", line 918, in __init__
    w.start()
  File "/usr/lib/python3.6/multiprocessing/process.py", line 105, in start
    self._popen = self._Popen(self)
  File "/usr/lib/python3.6/multiprocessing/context.py", line 223, in _Popen
    return _default_context.get_context().Process._Popen(process_obj)
  File "/usr/lib/python3.6/multiprocessing/context.py", line 277, in _Popen
    return Popen(process_obj)
  File "/usr/lib/python3.6/multiprocessing/popen_fork.py", line 19, in __init__
    self._launch(process_obj)
  File "/usr/lib/python3.6/multiprocessing/popen_fork.py", line 66, in _launch
    self.pid = os.fork()
OSError: [Errno 12] Cannot allocate memory
root@en-desktop:/jetson-inference/python/training/classification#    

Python version 3.6.9 installed on Jetson Nano Development Kit 4 gig

This is the command line start:

root@en-desktop:/jetson-inference/python/training/classification# python3 train.py --model-dir=models/cat_dog --batch-size=4 --workers=1 --epochs=1 data/cat_dog

I'm still looking at what the problem may be but any help would be appreciated.

Eric

dusty-nv, 2021-12-17:
> OSError: [Errno 12] Cannot allocate memory

Hi @enigma460, your board has run out of memory.  Please run with `--batch-size=1` instead and try these suggestions to mount SWAP/disable ZRAM and if needed disable the desktop GUI:

* https://github.com/dusty-nv/jetson-inference/blob/master/docs/pytorch-transfer-learning.md#mounting-swap
* https://github.com/dusty-nv/jetson-inference/blob/master/docs/pytorch-transfer-learning.md#disabling-the-desktop-gui

enigma460, 2021-12-17:
Thanks for the quick reply.
I was watching the memory used. I'm also using the minimum GUI in which at startup is only using 700 megs.
When running your demo I get 3.4 gig being used and 1,3  gig swap being used.
I also set the batch =  1
No joy.

enigma460, 2021-12-17:
Ok, After further experimentation and after raising the swap size to 8 gig, it finally completed without errors.

closed 2023-04-27 by dusty-nv