dusty-nv/jetson-inference #1333: Segmentation Fault (Core Dumped) Error while training

CiaranMcDermott, 2022-01-06:
Hi Dusty, 
I am following your tutorial on training an object detector on the Jetson nano. When I run the training command:

```python3 train_ssd.py --data=data/fruit --model-dir=models/fruit --batch-size=4 --epochs=3```

It gives the following error:

```Segmentation fault (core dumped)```

Nothing else is printed to the screen at all, so I'm assuming it doesn't even begin the training process. I have searched online and in other issues here and have found no solution that works for me. I have a 6 GB  swap file, so memory isn't the issue. I ran tegrastats while it was running and saw that memory was not an issue. I also tried disabling the desktop GUI but no luck.

I tried re-downloading the files from github using ```git clone --recursive https://github.com/dusty-nv/jetson-inference``` but the issue remained. 

I am using Jetpack version 4.6 and python 3.6. Using the docker container is not an option for me as it doesn't suit my project.
I am at a loss and have run out of ideas. If you have any suggestions at all they would be greatly appreciated.

Thanks in advance,

Ciaran

dusty-nv, 2022-01-06:
Hi @CiaranMcDermott, are you able to run `python3 -c 'import numpy'` ?

If not, can you try running `export OPENBLAS_CORETYPE=ARMV8` first?
https://github.com/numpy/numpy/issues/18131#issuecomment-756140369

CiaranMcDermott, 2022-01-06:
Hi @dusty-nv , 
Thanks for your prompt reply. 

Yes, numpy imports fine. There must be something else causing the error.

Ciaran

lacie-life, 2022-03-20:
I have same problem in Jetson AGX Xavier. Have any solution?

CiaranMcDermott, 2022-03-20:
Hi @lacie-life, I reinstalled the OS from scratch and didn't install anything else. It worked for me then.

Ciarán

closed 2022-03-20 by CiaranMcDermott

lacie-life, 2022-03-20:
I found a solution in issue ![https://github.com/dusty-nv/jetson-inference/issues/1114#issuecomment-925489693]. It worked.

maaaxac, 2024-04-09:
@lacie-life 
can you tell what was the solution for the issue? the url you showed is not working

lacie-life, 2024-04-10:
It's here: https://github.com/dusty-nv/jetson-inference/issues/1114#issuecomment-925489693

maaaxac, 2024-04-11:
Thank you, it worked!