dusty-nv/jetson-inference #1335: Dynamically train the model after deployment

irshadkn, 2022-01-20:
Is it possible to enhance the detection accuracy of the model after deployment by collecting datasets from the field and automatically training the model without manual intervention? The idea is to make the system learn without human intervention so that, the accuracy of the object detection increases day by day.

dusty-nv, 2022-01-20:
Hi @irshadkn, I haven't done this myself, but in theory yes it is possible.  You would have to experiment between two ways I think to see what worked better:

option 1.   add the new data to your existing dataset and train a completely new model on it each time
option 2.  fine-tune the model by making a new dataset from each set of new data, and load the previous model using the `--resume` or `--pretrained_ssd` arguments to train_ssd.py. 

Also, when you re-export the ONNX, remember to delete the `.engine` file in your model's directory so that TensorRT re-generates it.  Otherwise TensorRT will be using the old model.

irshadkn, 2022-01-21:
But, in this case, we have to manually annotate the data. I am looking for some method that collects its own dataset and trains itself without any human intervention. Something like reinforcement learning.

closed 2023-04-27 by dusty-nv