dusty-nv/jetson-inference #1337: Adjusting exposure with an open pipe line

jgan025, 2022-01-21:
Hi there,

I am currently having an issue where I am using a camera (with jetson) outdoors, and therefore need to account for very bright and very dark environments as the lighting changes natuerally. Using the auto exposure typically leads to oversaturation or dark regions of the object I am focussing on.

Reading previous issues, I can adjust the exposuretimerange and exposurecompensation to achieve satisfactory results, however this requires me to stop the pipe line, make the adjustment to the /utils/camera/gstCamera.cpp file and re-makeing.

**Is there any way to adjust the nvarguscamerasrc setting live with the pipeline open within my python script?** 

The goal being to bracket capture images with a range of expsores and then tonemap them to account for the dynamic range in the image. 

Thanks in advance!
Ming

dusty-nv, 2022-01-21:
Hi @jgan025, I'm not sure specifically about adjusting the exposure settings live, so you may want to ask that question on the Jetson forums.  At minimum it would require you to add additional code and Python bindings to set the GStreamer nvarguscamerasrc settings dynamically.

CameronTrumpy, 2022-01-21:
I have an extremely similar question. I have been working on a project within the jetson-inference docker container for a while( I haven't built the project from source), and have a sufficient understanding of the python layer, but don't understand the process of re-making python bindings. 

I have a specific USB camera I'd like to use, but when using it in jetson-inference, the autoexposure seems to set itself extremely dark. 

Since there seem to be no available python bindings to easily set exposure settings, I would like to edit the gstCamera::buildLaunchStr(), and either hard code in the gstreamer exposure or add a python binding to set it.

**I am struggling on the make step. Is there a way I can run make be run specifically on gstCamera.cpp? or does the make process require recompiling the whole project.**

Thanks a ton!
Cameron

dusty-nv, 2022-01-21:
> does the make process require recompiling the whole project.

The first time you build it, it will probably build the whole project.   But from then it doesn't need to re-build everything.   See the building from source instructions here:

https://github.com/dusty-nv/jetson-inference/blob/master/docs/building-repo-2.md#quick-reference

Remember to do the `sudo make install` step after you rebuild, otherwise it could use old binaries on the system.

jgan025, 2022-01-24:
> Hi @jgan025, I'm not sure specifically about adjusting the exposure settings live, so you may want to ask that question on the Jetson forums. At minimum it would require you to add additional code and Python bindings to set the GStreamer nvarguscamerasrc settings dynamically.

Appreciate the prompt response! I will try track down an answer, as surely someone has done this before. If I find something I'll mention it here.

closed 2023-04-27 by dusty-nv