dusty-nv/jetson-inference #1339: Re-training SSD-Mobilenet - loss going up and down

Ufosek, 2022-01-25:
Hi,

I am using transfer learning with Re-training SSD-Mobilenet like [here](https://github.com/dusty-nv/jetson-inference/blob/master/docs/pytorch-collect-detection.md).
My dataset contains 8000+ images (annotated sport players) (I have grayscale camera so all images are in grayscale).

I see that until 100 epochs loss is going down but then it is spiking and after exactly 200 epochs reaches new minimum.
1) I am wondering what does it mean (overfitting? or maybe that's just normal optimization)?
2) After each spike there is new minimum (100 - 1.47, 300 - 1.41, 500 - 1.39, 700 - 1.38) - Which one should I use? The lowest (at 700)? or at 100 (because later it may actually be not improving or even breaking)?

[image]

I would be glad for some help!
Regards

dusty-nv, 2022-01-25:
Hi @Ufosek, unfortunately this isn't exactly my area of specialty with interpreting such matters with training, so you may want to ask the same question on the upstream repo of pytorch-ssd:   https://github.com/qfgaohao/pytorch-ssd

What I would do is test both the epoch 100 model and epoch 700 model and see which one works better in practice for your application.

Ufosek, 2022-01-25:
Ok thx!