dusty-nv/jetson-inference #1340: Stereo Depth

mosimp, 2022-01-25:
Hi Dusty,

It looks like you've done some experimentation with Stereo Depth.  Is adding Stereo Depth on the Jetson-Inference roadmap?  
From what you've seen, does running a Stereo Depth DNN on the Jetson Nano seem feasible?

Thanks!

dusty-nv, 2022-01-25:
Hi @mosimp, I don't currently have active plans to move it out of the experimental branch, but you can also find it here:  https://docs.nvidia.com/isaac/archive/2020.1/packages/stereo_depth/doc/sddnn.html

Part of the thing is, GPU-accelerated stereo depth estimation is also in VPI now and many stereo cameras come with their own stereo depth implementations (like ZED SDK).   Whereas with monocular depth estimation, that is pretty much totally done with DNNs.

mosimp, 2022-01-25:
Makes sense.  Thank you!

kiplimock, 2022-03-31:
Hi @dusty-nv, 

Just a quick follow-up on the second part of @mosimp's question. Is it possible to run Stereo Depth DNN on a Jetson Nano?

dusty-nv, 2022-03-31:
The size of the stereo DNN (at least the one I had played around with in this repo) makes it unrealistic to deploy on Nano.   I would look at the ZED SDK instead...I think they have a more optimized stereo DNN now.