dusty-nv/jetson-inference #1342: Changing of "boxes"

qasmendoza, 2022-01-26:
Hi @dusty-nv, how can I change these boxes from "filled" with color (green for instance) to just the boxes alone? Thank you in advance.

qasmendoza, 2022-02-01:
@dusty-nv any inputs? :)

qasmendoza, 2022-02-01:
anyone? a little help?

dusty-nv, 2022-02-01:
Hi @qasmendoza, I just added support for this in https://github.com/dusty-nv/jetson-inference/commit/cf9376ff9a78dcd0f512d90e0ea66c37f757261b

You can now run `detectnet --overlay=lines` to get just the lines  (or `--overlay=lines,labels,conf` for example)

qasmendoza, 2022-02-06:
hi @dusty-nv it didn't work. I've used detectnet --overlay=lines and it didn't detect anything. I deleted --overlay=lines and it can detect the object but still on green boxes.

dusty-nv, 2022-02-07:
> hi @dusty-nv it didn't work. I've used detectnet --overlay=lines and it didn't detect anything. I deleted --overlay=lines and it can detect the object but still on green boxes.

Did you update your repo or clone the latest source from master and rebuild/reinstall?

qasmendoza, 2022-02-07:
Sorry I did not. Let me try that and thanks for clarification.

closed 2023-04-27 by dusty-nv