dusty-nv/jetson-inference #1344: how to use .pth training in python code or convert .pth to .uff

SachaMk2, 2022-01-29:
Hi,

I'm using your github repository (jetson-inference) for an object detection project. I used a python code to run mobilenet-ssd in realtime (100% based on your tutorial and this one https://www.youtube.com/watch?v=mB025B7KpeE)

The code works perfectly but I would like to try and make it work with my model trained on my own dataset images.

So I followed your tutorial which also worked very well but my problem is that the output is a .pth (or onnx if converted) and unfortunately when I go to "jetson-inference/data/mobilenetssd-v2" I see that there are 2 files (not including labels): one named "ssd_mobilenet_v2_coco.uff" and another "sdd_mobilenet_v2_coco.uff.1 .1.8001.GPU.FP16.engine
Could you tell me how I can do the training on my data and have as a result this output. Or if there is a way to convert pth into uff ?

Or how do i use a .pth result of training in my python code.

this is my code:

        `import jetson.inference
        import jetson.utils
        import cv2
  
  
        class mnSSD():
	        def __init__(self, path, threshold):
		        self.path = path
		        self.threshold = threshold
		        self.net = jetson.inference.detectNet(self.path, self.threshold)
		        
	        def detect(self, img, display=False):
		        imgCuda = jetson.utils.cudaFromNumpy(img)
		        detections = self.net.Detect(imgCuda, overlay = "OVERLAY_NONE")
		        
		        objects=[]
		        
		        for d in detections:
			        className = self.net.GetClassDesc(d.ClassID)
			        objects.append([className, d])
			        #print(d)
			        if display:
				        x1, y1, x2, y2 = int(d.Left), int(d.Top), int(d.Right), int(d.Bottom)
				        cx, cy = int(d.Center[0]), int(d.Center[1])
				        cv2.rectangle(img, (x1,y1), (x2,y2), (255,0,255), 2)
				        cv2.line(img, (x1,cy), (x2, cy), (255,0,255), 1)
				        cv2.line(img, (cx,y1), (cx, y2), (255,0,255), 1)
				        cv2.circle(img, (cx, cy), 5, (255,0,255), cv2.FILLED)
				        cv2.putText(img, className, (x1+5, y1+15), cv2.FONT_HERSHEY_DUPLEX, 0.75, (255,0,255), 2)
				        cv2.putText(img, f'FPS: {int(self.net.GetNetworkFPS())}', (30,30), cv2.FONT_HERSHEY_DUPLEX, 1, (255,0,0), 2)
		        
		        
		        return objects
		  
  
  
	  
	  
	  
          def main():
	          
	          
	          #cap = cv2.VideoCapture(gstreamer_pipeline(flip_method=0), cv2.CAP_GSTREAMER)
	          cap = cv2.VideoCapture(0)
	          cap.set(3, 640)
	          cap.set(4, 480)
	          myModel = mnSSD("ssd-mobilenet-v2", 0.5)
	          while True:
		          success, img = cap.read()
		          objects = myModel.detect(img, True)
		          if len(objects) !=0:
			          print(objects[0][0])
			          
		          cv2.imshow("Image", img)
          
		          cv2.waitKey(1)
	          
          if __name__ == "__main__":
	          main()

`

Thanks in advance for your help

Best regards

dusty-nv, 2022-01-29:
That UFF ssd-mobilenet is a different model that comes with the repo.

Instead run detectnet with the paths to your onnx model, like the command-line arguments shown here:

https://github.com/dusty-nv/jetson-inference/blob/master/docs/pytorch-ssd.md#processing-images-with-tensorrt

SachaMk2, 2022-01-29:
hi,

thanks for your answer.

An other question, how can i train UFF ssd-mobilenet or  how can i use detectnet in my python code.

Thanks in advance

dusty-nv, 2022-01-31:
Hi @SachaMk2, the UFF model came from TensorFlow and this repo to convert it to UFF:  https://github.com/AastaNV/TRT_object_detection

The ONNX and PyTorch is the recommended way now to train/deploy SSD-Mobilenet with detectnet.   If you want to load a custom ONNX model with detectNet in your Python code, you can do it like shown here:

https://github.com/dusty-nv/jetson-inference/issues/786#issuecomment-721973736

closed 2023-04-27 by dusty-nv