dusty-nv/jetson-inference #1345: Jetson Xavier AGX Dev kit no internet, no bluetooth

tulbureandreit, 2022-01-31:
Hello,

A quick question. I bought a Jetson Xavier AGX Dev kit and I did not use it for about 2-3 months.

Now, I try to use it and to my surprise, OS and jetpack was already installed, so no need to flash it.

Moreover, on startup, I see a line that says Wifi and bluetooth script init failed.

And Ethernet connection is also not working properly hence I do not have any internet on my board.

Any ideas on how to fix it ? I really need my board by Friday.

Thank you!

dusty-nv, 2022-01-31:
Hi @tulbureandreit, since your question is unrelated to this repo, please post it to the Jetson AGX Xavier forum:

https://forums.developer.nvidia.com/c/agx-autonomous-machines/jetson-embedded-systems/jetson-agx-xavier/75

Wish you the best of luck in getting the networking going!   My suggestion is probably to re-flash it with the latest JetPack.

closed 2022-01-31 by dusty-nv

tulbureandreit, 2022-01-31:
Okay thank you !
Moreover, I see that I cannot flash it from a Host computer with Ubuntu20.04. So do I need to install 18.04 on one?
Kind regards, Toate cele bune,
Tulbure Andrei

 
 

Closed #1345.