dusty-nv/jetson-inference #1346: Convert gstCamera float to OpenCV Mat

aviontics, 2022-02-02:
here is link to last issue
https://github.com/dusty-nv/jetson-inference/issues/361#issuecomment-521428329

I have had this issue in 2019 and was resolved using the    ConvertRGBA function in the gstcamera.cpp file.

     if (!camera->ConvertRGBA(imgCUDA, &imgRGBA))
        {
            printf("detectnet-camera 0:  failed to convert from NV12 to RGBA\n");
        }
     im = cv::Mat(cvheight, cvwidth, CV_8UC3, &imgCPU);

The function was removed in mid 2020. 

I have tried using camera->CaptureRGBA(&imgRGBA, 1000, true)
and    im = cv::Mat(cvheight, cvwidth, CV_8UC3, imgRGBA);
The retrun image seems like gibberish pixels before crashing.

Is there a simple way to convert from either C++ gstcamera float or uchar3* image to OpenCV MAT?

dusty-nv, 2022-02-02:
Hi @aviontics, it's recommended to migrate to the `videoSource::Capture()` interface instead:

https://github.com/dusty-nv/jetson-inference/blob/master/docs/aux-streaming.md#c

As shown in that example, this will give you a uchar3 image instead of float
It's actually templated, so it can give you uchar3/uchar4/float3/float4 (whichever datatype you prefer)

aviontics, 2022-02-02:
> Hi @aviontics, it's recommended to migrate to the `videoSource::Capture()` interface instead:
>
> https://github.com/dusty-nv/jetson-inference/blob/master/docs/aux-streaming.md#c
>
> As shown in that [...]

Thanks for the prompt reply. I have modified code to use videoSource::Capture()

Is there a function to convert to opencv MAT format or should just pass the unchar3 image into the opencv MAT variable like this?
 im = cv::Mat(cvheight, cvwidth, CV_8UC3, unchar3 image);

The python opencv example conversion exists but its missing so far for C++

dusty-nv, 2022-02-02:
> Is there a function to convert to opencv MAT format or should just pass the unchar3 image into the opencv MAT variable like this?
> im = cv::Mat(cvheight, cvwidth, CV_8UC3, unchar3 image);

I'm not super familiar myself, but that is the way I would try first.

aviontics, 2022-02-02:
> > Is there a function to convert to opencv MAT format or should just pass the unchar3 image into the opencv MAT variable like this?
> > im = cv::Mat(cvheight, cvwidth, CV_8UC3, unchar3 image);
> 
> I'm not super familiar myself, but that is the way I would try first.

I tried and its failed as it requires void* type in  cv::Mat(cvheight, cvwidth, CV_8UC3, (void)* image);

Think may have something to do with the memory allocated to store image read.

Issue I have is that there does not seem to interaction between image data in uchar3* with the void* type data created, like void* imgCUDA = NULL;.

In the previous versions that interaction happened in this function;
bool Capture( void** cpu, void** cuda, uint64_t timeout=UINT64_MAX );

as in use case below

 if (!camera->Capture(&imgCPU, &imgCUDA, 1000))

Is there any such function?

dusty-nv, 2022-02-02:
Does `cv::Mat(cvheight, cvwidth, CV_8UC3, (void)* image);` not work when you cast it to `void*` there?

You should just be able to cast the pointer to `void*` when you need it to be `void*`

> as in use case below
> 
> if (!camera->Capture(&imgCPU, &imgCUDA, 1000))
> 
> Is there any such function?

Yes, you can use this overload with `IMAGE_RGB8`:  https://github.com/dusty-nv/jetson-utils/blob/6f0c5b74b59e08f8205d95f19c42d9b20aa79112/video/videoSource.h#L214

aviontics, 2022-02-04:
> Does `cv::Mat(cvheight, cvwidth, CV_8UC3, (void)* image);` not work when you cast it to `void*` there?
>
> You should just be able to cast the pointer to `void*` when you need it to be `void*`
>
>
> Yes, [...]

I got it working using:

    uchar3* imgRGBA = NULL;

    while (!signal_recieved)
    {

                if (!camera->Capture(&imgRGBA, 1000))
        {
            printf("detectnet-camera 0:  failed to convert from NV12 to RGBA\n");
        }
             CUDA(cudaDeviceSynchronize()); 
     im = cv::Mat(cvheight, cvwidth, CV_8UC3, (void*) imgRGBA);

Thank for your help.

closed 2023-04-27 by dusty-nv