dusty-nv/jetson-inference #1349: Jetpack 4.3 generated mobilenet.uff file working differently in Jetpack 4.6

aviontics, 2022-02-04:
I generated a modilenet .uff file in jetpack 4.3 for my custom mobilenet model and when ever I switched to a different TX2 with same jetpack 4.3 a new Tensorrt engine file was generated for that new TX2. All work same and fine in terms of detections. 

However when I repeat the steps in Jetson NX with Jetpack 4.6, the Jetson Inference generates a new Tensorrt engine file _ssd_mobilenet_v2_coco.uff.1.1.8001.GPU.FP16.engine_ from the old _ssd_mobilenet_v2_coco.uff file_ .  However the detection is different and seems to act differently from the custom mobilenet TensorRT engine generated in TX2 Jetpack 4.3.  This may have something to do with versions of Tensorrt or Tensorflow.

Question is, which is it?  Is it the TensorRT version or the Tensorflow version that is causing the issue in  Jetpack 4.6.

Jetpack has these versions:
tensorboard-plugin-wit        1.8.1
tensorflow                    2.6.2+nv21.12
tensorflow-estimator          2.6.0
tensorrt                      8.0.1.6

Jetpack 4.3 Version
tensorrt                      6.0.1.10
uff                           0.6.5

Notice that tensorflow is not present in the Jetpack 4.3 TX2. 

By deduction I would think its the TensorRT version difference that affected the way the mobilenet.uff is used to generate the TensorRT engine file.

Am I correct? If so, how do I fix this issue?

dusty-nv, 2022-02-04:
Hi @aviontics, are you saying that your same mobilenet.uff behaves differently on JetPack 4.3 vs JetPack 4.6?

aviontics, 2022-02-04:
> Hi @aviontics, are you saying that your same mobilenet.uff behaves differently on JetPack 4.3 vs JetPack 4.6?

Yes, exactly. The inference engine is generated from the .uff file from Jetpack 4.3, without any clear error and the video streams fine but the detection is different. 

I do not understand how this is possible.

dusty-nv, 2022-02-04:
Hmm I am not sure either, sorry.  It sounds related to newer version of TensorRT?

FWIW I don't use UFF much, primarily I use ONNX.  The only UFF model I use is that ssd-mobilenet-coco one, and I haven't changed it in some time.  Going forward I am always using ONNX.

aviontics, 2022-02-04:
> 

Is it possible to use an older TensorRT in Jetpack 4.6 or I have to downgrade to older Jetpack like 4.4.

dusty-nv, 2022-02-04:
You would need to downgrade to an older version of JetPack.

closed 2023-04-27 by dusty-nv