dusty-nv/jetson-inference #1350: How to count the total number of objects

shi093, 2022-02-05:
Hi, Dustin,
Thank you so much for offering the Hello AI World tutorials. After following the tutorial "Locating Objects with DetectNet", I am wondering how I could count the total number of objects in each image (or video frame). Could you please direct me to the resources that I could use to build on this jetson-inference library and do more customizable counting?
Thanks a lot.

dusty-nv, 2022-02-07:
Hi @shi093, there is documentation on the detections structures that gets returned by `detectNet.Detect()` on this page:  

https://rawgit.com/dusty-nv/jetson-inference/dev/docs/html/python/jetson.inference.html#detectNet

```
Detection = <type 'jetson.inference.detectNet.Detection'>
Object Detection Result
 
----------------------------------------------------------------------
Data descriptors defined here:
 
Area
    Area of bounding box
 
Bottom
    Bottom bounding box coordinate
 
Center
    Center (x,y) coordinate of bounding box
 
ClassID
    Class index of the detected [object](https://rawgit.com/dusty-nv/jetson-inference/dev/docs/html/python/__builtin__.html#object)
 
Confidence
    Confidence value of the detected [object](https://rawgit.com/dusty-nv/jetson-inference/dev/docs/html/python/__builtin__.html#object)
 
Height
    Height of bounding box
 
Instance
    Instance index of the detected [object](https://rawgit.com/dusty-nv/jetson-inference/dev/docs/html/python/__builtin__.html#object)
 
Left
    Left bounding box coordinate
 
Right
    Right bounding box coordinate
 
Top
    Top bounding box coordinate
 
Width
     Width of bounding box
```

You can use this information to build customized statistics and counting of the objects.   Or if you simply want to know the total number of detected objects by the DNN, just do `len(detections)`

shi093, 2022-03-07:
Dusty, thank you so much for answering my question, really appreciated. Now I am wondering how I could move onto object tracking, i.e., give each detected object a unique ID and track it from frame to frame. I saw some tutorials on object tracking using openCV, but I would like to build the object tracking program on top of the object detection model I re-trained using jetson-inference library. Could you please point me to the right direction for doing this? Thanks again.

dusty-nv, 2022-03-07:
Hi @shi093, I don't have tracking functionality in jetson.inference, however DeepStream does and there are ways to use the trained ssd-mobilenet with DeepStream:   https://elinux.org/index.php?title=Jetson/L4T/TRT_Customized_Example#Custom_Parser_for_SSD-MobileNet_Trained_by_Jetson-inference

You can also convert the image to numpy array and use the openCV functions for tracking:   https://github.com/dusty-nv/jetson-inference/blob/master/docs/aux-image.md#converting-to-numpy-arrays

https://github.com/dusty-nv/jetson-utils/blob/7de695d94271ba1ae77a4358f427c0887b09084f/python/examples/cuda-to-cv.py

rajeshroy402, 2022-04-05:
@dusty-nv  - I have a question regarding the mobilenet ssd v2 parser for deepstream, it says that there should be a .uff file, but we don't have .uff file right?
 [I am having a custom model trained]

dusty-nv, 2022-04-05:
The 90-class ssd-mobilenet-v2 that comes with jetson-inference is from TensorFlow (UFF)

However, the custom-trained ssd-mobilenet-v1 models that are trained with PyTorch are ONNX

closed 2023-04-27 by dusty-nv