dusty-nv/jetson-inference #1351: Prediction result automatically closes on Nvidia jetson xavier AGX

tulbureandreit, 2022-02-05:
Hello,

I tried to run the image recognition tutorial and when I run 

./imagenet.py images/orange_0.jpg images/test/output_0.jpg

The system starts processing and then the output windows automatically closes. Any idea why?

rajeshroy402, 2022-02-06:
Hi @tulbureandreit , 
Can you share some more details on the hardware which you are using and which JP is it?

Also, is the output image getting saved in your `images/test/output_0.jpg`

The most relatable answer to your query is that since you are just processing 1 input image, the tiler/display window opens up and closes after processing the inference.

Thanks

dusty-nv, 2022-02-07:
> The most relatable answer to your query is that since you are just processing 1 input image, the tiler/display window opens up and closes after processing the inference.

Thanks @rajeshroy402, I also believe this to be the correct answer.

If you want it to stay open forever, you can keep looping the image with `--input-loop=-1` 
https://github.com/dusty-nv/jetson-inference/blob/master/docs/aux-streaming.md#input-options

closed 2023-04-27 by dusty-nv