dusty-nv/jetson-inference #1352: How can I use object detection for occupancy detection

jumill87, 2022-02-07:
Hello, I am creating a new project that is detecting empty chairs vs occupied chairs and need any suggestions/help or guidance on what to do.  So far, I've trained a model/taken pictures of empty chairs and chairs with people in them and have it only detecting them with my dataset/labels.  

However, I saw a project where it counts people crossing a line and crossing back, giving the people numbers instead of detecting them as a person, is there a way to morph this idea into detecting when a chair is occupied or not?  Can I make a bounding box around the empty chair and when a person enters it have it detect that its occupied?  Any guidance/help would be appreciated, having a rough time trying to figure this out.

Thank you.

dusty-nv, 2022-02-07:
Hi @jumill87, DeepStream SDK can do things like that, with boundary zones and such and people counting.  So you might want to check out those samples too.

But in theory with the logic you outlined, you should be able to do it with jetson.inference too.  jetson.inference does not have tracking however, which makes frame-to-frame analysis like this more difficult.  DeepStream does have tracking.

Is your camera stationary?  That would make it a lot easier

jumill87, 2022-02-08:
Currently we are using the logitech USB webcam and keeping it stationary ( we actually plan on encasing it (the jetson) in one of those metal waveshare cases with the camera mounting bracket on top ).  Tracking the chairs is not our biggest concern as of now we just want to make sure we can see the chairs empty/occupied.  I will check DeepStream out and report back if I find anything out or manage to figure something out.  Thank you for the advice!

Thank you so much for replying Dusty! I love your tutorials and really appreciate everything you do for the community.  So far we are going to just use the models we have and make our program subtract the number of empty chairs from the chairs taken to give us an output of how many seats are available for our webserver to keep track of and hopefully that all works out.  When I followed your tutorials and got everything working it blew my mind, really amazing what these devices can do.

dusty-nv, 2022-02-08:
OK gotcha, yea if the camera is stationary, that certainly makes things a lot easier RE: the need for tracking.

Glad you have enjoyed the tutorials and found them useful!  :)

jumill87, 2022-02-25:
Hey Dusty, 

Thanks for the responses, again, I really appreciate it. 

I am having trouble installing Deepstream 6.0 on my Nano, I've followed all these instructions: 
https://docs.nvidia.com/metropolis/deepstream/dev-guide/text/DS_Quickstart.html

I managed to install the things it wants but when I try to run the sample apps I get errors, I'm really out of my depth here any guidance or info would be appreciated, should I do a fresh install of JetPack and then try installing it again?  Maybe something I've installed previously is conflicting with it?  I have no idea, if you don't know thanks again for taking the time to help anyways.  

Sincerely,

Justin

dusty-nv, 2022-02-25:
Hi @jumill87, have you tried the [deepstream-l4t](https://catalog.ngc.nvidia.com/orgs/nvidia/containers/deepstream-l4t) container?

If you continue having problems, I would reflash the device and if needed post to the [Deepstream SDK Forum](https://forums.developer.nvidia.com/c/accelerated-computing/intelligent-video-analytics/deepstream-sdk/15)

jumill87, 2022-02-28:
Hey @dusty-nv, I got it working now.  There is actually a course NVIDIA offers to get you started with learning it, really awesome.

I had one last question, you may or may not have an answer for me and that's ok.  

Is there anyway to take the data or make a database from what my object detection model is detecting?  Like for example, say I'm monitoring the occupancy of my chairs, and I want to save that data or upload it to a webserver to make a website that will display how many chairs are empty/taken.  Is that even possible at the moment? Or did I get way ahead of myself with this project (lol)? 

Basically we want to create a website that will update live based on what our object detection model is detecting, like if a chair is taken on the website it updates to say chair taken, when someone leaves it, it updates to chair empty.  I don't know if I'm being clear, I've scoured the internet and cannot find anything other than using some type of text messaging service to notify people how many chairs are left in a room vs total amount of chairs.  Again, really appreciate all the help.  If anyone else wants to know about the course for deepstream, here is the link: [](https://courses.nvidia.com/courses/course-v1:DLI+S-IV-02+V2/course/)

dusty-nv, 2022-02-28:
Sure thing, you can do that, although that is more of a web development question than it is related to Jetson specifically.   For example, you could use a lightweight [sqlite3](https://docs.python.org/3/library/sqlite3.html) database and then make a web frontend for it (i.e. with Python flask or dash or some other web framework).  It's kind of an independent question from the computer vision / inferencing component and is outside of my area of expertise.

DeepStream also supports a bunch of different message brokers like Kafka, MQTT, AQMP, ect which you can use to build edge-to-cloud applications that are more robust.

closed 2023-04-27 by dusty-nv