dusty-nv/jetson-inference #1353: following your instructions for collecting my own object detection models doesn't work

Gianni2045, 2022-02-07:
2022-02-07 08:30:40 - Using CUDA...
2022-02-07 08:30:40 - Namespace(balance_data=False, base_net=None, base_net_lr=0.001, batch_size=4, checkpoint_folder='models/Cosmetics', dataset_type='voc', datasets=['data/Cosmetics'], debug_steps=10, extra_layers_lr=None, freeze_base_net=False, freeze_net=False, gamma=0.1, lr=0.01, mb2_width_mult=1.0, milestones='80,100', momentum=0.9, net='mb1-ssd', num_epochs=100, num_workers=2, pretrained_ssd='models/mobilenet-v1-ssd-mp-0_675.pth', resume=None, scheduler='cosine', t_max=100, use_cuda=True, validation_epochs=1, weight_decay=0.0005)
2022-02-07 08:30:40 - Prepare training datasets.
Traceback (most recent call last):
  File "train_ssd.py", line 214, in <module>
    target_transform=target_transform)
  File "/jetson-inference/python/training/detection/ssd/vision/datasets/voc_dataset.py", line 36, in __init__
    self.ids = self._read_image_ids(image_sets_file)
  File "/jetson-inference/python/training/detection/ssd/vision/datasets/voc_dataset.py", line 111, in _read_image_ids
    if self._get_num_annotations(image_id) > 0:
  File "/jetson-inference/python/training/detection/ssd/vision/datasets/voc_dataset.py", line 123, in _get_num_annotations
    objects = ET.parse(annotation_file).findall("object")
  File "/usr/lib/python3.6/xml/etree/ElementTree.py", line 1196, in parse
    tree.parse(source, parser)
  File "/usr/lib/python3.6/xml/etree/ElementTree.py", line 586, in parse
    source = open(source, "rb")
ValueError: embedded null byte

Why am I getting these errors, I spent hours taking pictures for horrible results
The only difference I did was not to press the merge button I wanted to learn the regular way

Please help me ASAP I have spent countless hours trying to figure out what the problem is and there is no information on github or nvidia to help me
[labels.txt](https://github.com/dusty-nv/jetson-inference/files/8013585/labels.txt)
[labels.txt](https://github.com/dusty-nv/jetson-inference/files/8013595/labels.txt)

Gianni2045, 2022-02-07:
2022-02-07 08:30:40 - Using CUDA...
2022-02-07 08:30:40 - Namespace(balance_data=False, base_net=None, base_net_lr=0.001, batch_size=4, checkpoint_folder='models/Cosmetics', dataset_type='voc', datasets=['data/Cosmetics'], debug_steps=10, extra_layers_lr=None, freeze_base_net=False, freeze_net=False, gamma=0.1, lr=0.01, mb2_width_mult=1.0, milestones='80,100', momentum=0.9, net='mb1-ssd', num_epochs=100, num_workers=2, pretrained_ssd='models/mobilenet-v1-ssd-mp-0_675.pth', resume=None, scheduler='cosine', t_max=100, use_cuda=True, validation_epochs=1, weight_decay=0.0005)
2022-02-07 08:30:40 - Prepare training datasets.
Traceback (most recent call last):
File "train_ssd.py", line 214, in
target_transform=target_transform)
File "/jetson-inference/python/training/detection/ssd/vision/datasets/voc_dataset.py", line 36, in init
self.ids = self._read_image_ids(image_sets_file)
File "/jetson-inference/python/training/detection/ssd/vision/datasets/voc_dataset.py", line 111, in _read_image_ids
if self._get_num_annotations(image_id) > 0:
File "/jetson-inference/python/training/detection/ssd/vision/datasets/voc_dataset.py", line 123, in _get_num_annotations
objects = ET.parse(annotation_file).findall("object")
File "/usr/lib/python3.6/xml/etree/ElementTree.py", line 1196, in parse
tree.parse(source, parser)
File "/usr/lib/python3.6/xml/etree/ElementTree.py", line 586, in parse
source = open(source, "rb")
ValueError: embedded null byte

Why am I getting these errors, I spent hours taking pictures for horrible results
The only difference I did was not to press the merge button I wanted to learn the regular way

Please help me ASAP I have spent countless hours trying to figure out what the problem is and there is no information on github or nvidia to help me
[labels.txt](https://github.com/dusty-nv/jetson-inference/files/8013585/labels.txt)

Gianni2045, 2022-02-07:
I highly respect you and I know you will get me out of this with the help I need

voidzer0-development, 2022-02-07:
You should go to https://github.com/qfgaohao/pytorch-ssd which is where the training code is branched from and see if someone had a similar error

Gianni2045, 2022-02-07:
thank you very much I hope dusty gets back to me soon please and thank you

Gianni2045, 2022-02-08:
thank you sir you are an admirable worker

Gianni2045, 2022-02-08:
thank you very much you truly are the best teacher I ever had

dusty-nv, 2022-02-08:
No problem @Gianni2045, happy to help :)

Hope you get it working, if not let me know.

Gianni2045, 2022-02-09:
please forgive my ignorance i try my best not to be but I have been trying to send you my information attaching my files here it wont accept that type of file, how do I please send you the data in google drive?

GianniM2045, 2022-02-12:
[20220208-022651.zip](https://github.com/dusty-nv/jetson-inference/files/8052851/20220208-022651.zip)

GianniM2045, 2022-02-12:
I think I figured it out how to share my data thank you so much for your care as a teacher

GianniM2045, 2022-02-12:
hope i will understand your feedback on this shared data

Gianni2045, 2022-02-12:
I sent you the data please with all respect lete know what information of problems you found in my data Dusty

dusty-nv, 2022-02-14:
Hi @Gianni2045 @GianniM2045, I'm having trouble downloading all the individual folders you sent me on Google Drive.  Can you compress your entire dataset into a single .zip or .tar.gz archive and send me that on google drive instead?  Thanks.

GianniM2045, 2022-02-14:
Yes I will try

GianniM2045, 2022-02-15:
[20220208-022651.zip](https://github.com/dusty-nv/jetson-inference/files/8065725/20220208-022651.zip)

GianniM2045, 2022-02-15:
20220208-022651.zip
<https://drive.google.com/file/d/1QaMNDQ-UAQogwbqAomApEzW1D6OlN2V_/view?usp=drive_web>

Message ID: ***@***.***>
>>
>

dusty-nv, 2022-02-15:
> https://drive.google.com/file/d/1QaMNDQ-UAQogwbqAomApEzW1D6OlN2V_/view?usp=drive_web

Hi @GianniM2045, is this the link?  I've requested access to download it.

GianniM2045, 2022-02-15:
suddes.zip
<https://drive.google.com/file/d/1tgTWfsq-xUyL2iNOywVysq4vb2Et66o9/view?usp=drive_web>

GianniM2045, 2022-02-15:
yes this is the link and this one also

GianniM2045, 2022-02-15:
thank you very much

Gianni2045, 2022-02-23:
> suddes.zip
> https://drive.google.com/file/d/1tgTWfsq-xUyL2iNOywVysq4vb2Et66o9/view?usp=drive_web
>
> OK, I requested access to this one also

I accepted your request thank you so much for your help

closed 2023-04-27 by dusty-nv

GianniM2045, 2023-04-27:
Thank you so much for completing my submission

GianniM2045, 2023-04-27:
Hope you are doing well dusty, Jesus bless you always.