dusty-nv/jetson-inference #1355: Pose Estimation on Jetson Xavier NX

irshadkn, 2022-02-09:
I followed the below tutorial.
https://github.com/dusty-nv/jetson-inference/blob/master/docs/posenet.md

I tried the code from the section "Working with Object Poses" 

It resulted in the following error:

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "posenet_ex.py", line 33, in <module>
    left_shoulder_idx = pose.FindKeypoint('left_shoulder)
SystemError: <built-in method FindKeypoint of jetson.inference.poseNet.ObjectPose object at 0x7f7492ac30> returned a result with an error set

ptrxpl, 2022-02-10:
Hello,
I have the same problem (on TX2 + Kinect 2.0, but doesn't matter), but finally managed to get it:
1) You can use: 
```python 
net.FindKeypointID('left_shoulder')
```
`net` stands from this code: https://github.com/dusty-nv/jetson-inference/blob/master/python/examples/posenet.py

2) Also you can iterate over this:
```python
for keypoint in pose.Keypoints:
    print(keypoint.ID)
```

It should print IDs of actually detected keypoints on image.

Documentation is here: 
https://rawgit.com/dusty-nv/jetson-inference/dev/docs/html/python/jetson.inference.html#poseNet

dusty-nv, 2022-02-11:
Ah sorry for the typo!, thanks @ptrxpl @irshadkn 

Just fixed this in https://github.com/dusty-nv/jetson-inference/commit/91f0a7022ccb3bab07b1770cec812ab1c9da6138

ptrxpl, 2022-02-13:
Hey @dusty-nv
I think your fix isn't enough (there is no FindKeypoint, but FindKeypointID - but kinda useless right here), please consider following code (to update example on posenet.md):

```python
nose_idx = 0
left_eye_idx = 1
right_eye_idx = 2
left_ear_idx = 3
right_ear_idx = 4
left_shoulder_idx = 5
right_shoulder_idx = 6
left_elbow_idx = 7 
right_elbow_idx = 8
left_wrist_idx = 9
right_wrist_idx = 10
left_hip_idx = 11
right_hip_idx = 12
left_knee_idx = 13
right_knee_idx = 14
left_ankle_idx = 15
right_ankle_idx = 16
neck_idx = 17

for pose in poses:
        detected_kepoints = []
        detected_kepoints.extend([keypoint.ID for keypoint in pose.Keypoints])

        if all(x in detected_kepoints for x in [left_wrist_idx, left_shoulder_idx]):
            left_wrist = pose.Keypoints[left_wrist_idx]
            left_shoulder = pose.Keypoints[left_shoulder_idx]
            
            point_x = left_shoulder.x - left_wrist.x
            point_y = left_shoulder.y - left_wrist.y

            print(f"person {pose.ID} is pointing towards ({point_x}, {point_y})")
```

Also tagging @irshadkn

dusty-nv, 2022-02-14:
OK, sorry - in my haste I didn't realize that the original Python example code was fine and it was the poseNet Python bindings that had a bug.   I have fixed/verified the bindings and reverted the example code to the original (using `pose.FindKeypoint()`) in https://github.com/dusty-nv/jetson-inference/commit/b1afeb89b48224fbd33b420e675099bfa6a422f9

So if you pull/build/install the latest master, the original way should be working now.  @ptrxpl that saves you the step of manually looking up the keypoints by their ID in the list of pose keypoints (although your way works too)

closed 2023-04-27 by dusty-nv