dusty-nv/jetson-inference #1358: jetson nano(4G)：Script detectnet.py   stopped and that videoSource failed to capture image（using V4L2 camera）

tonylau2010, 2022-02-13:
Hi！
   Would you help me,if similar/same err is occured? Thanks s lot！

   When I try to detect objection using the maner as follows：
   https://github.com/dusty-nv/jetson-inference/blob/master/docs/detectnet-camera-2.md

   After that：  ./detectnet.py /dev/video0             # V4L2 camera

   The Log is as follows：（and the script is stopped，then the new window   is crash）
detected 1 objects in image
<detectNet.Detection object>
   -- ClassID: 1
   -- Confidence: 0.937988
   -- Left:    0.9375
   -- Top:     13.8281
   -- Right:   333.75
   -- Bottom:  476.484
   -- Width:   332.812
   -- Height:  462.656
   -- Area:    153978
   -- Center:  (167.344, 245.156)
[OpenGL] glDisplay -- set the window size to 640x480
[OpenGL] creating 640x480 texture (GL_RGB8 format, 921600 bytes)
[cuda]      unknown error (error 999) (hex 0x3E7)
[cuda]      /home/tony-nano/Downloads/readyZip/jetson-inference/utils/display/glTexture.cpp:360

[TRT]    ------------------------------------------------
[TRT]    Timing Report networks/SSD-Mobilenet-v2/ssd_mobilenet_v2_coco.uff
[TRT]    ------------------------------------------------
[TRT]    Pre-Process   CPU   0.10620ms  CUDA   0.95193ms
[TRT]    Network       CPU 4152.17236ms  CUDA 4150.15332ms
[TRT]    Post-Process  CPU   0.04583ms  CUDA   0.04625ms
[TRT]    Visualize     CPU  12.51637ms  CUDA  12.75078ms
[TRT]    Total         CPU 4164.84082ms  CUDA 4163.90283ms
[TRT]    ------------------------------------------------

[TRT]    note -- when processing a single image, run 'sudo jetson_clocks' before
                to disable DVFS for more accurate profiling/timing measurements

[cuda]      unknown error (error 999) (hex 0x3E7)
[cuda]      /home/tony-nano/Downloads/readyZip/jetson-inference/utils/cuda/cudaYUV-YUYV.cu:160
[cuda]      unknown error (error 999) (hex 0x3E7)
[cuda]      /home/tony-nano/Downloads/readyZip/jetson-inference/utils/cuda/cudaColorspace.cpp:75
[cuda]      unknown error (error 999) (hex 0x3E7)
[cuda]      /home/tony-nano/Downloads/readyZip/jetson-inference/utils/codec/gstBufferManager.cpp:383
[gstreamer] gstBufferManager -- unsupported image format (rgb8)
[gstreamer]                     supported formats are:
[gstreamer]                        * rgb8
[gstreamer]                        * rgba8
[gstreamer]                        * rgb32f
[gstreamer]                        * rgba32f
[gstreamer] gstDecoder -- failed to retrieve next image buffer
Traceback (most recent call last):
  File "./tt.py", line 63, in <module>
    img = input.Capture()
Exception: jetson.utils -- videoSource failed to capture image
[gstreamer] gstCamera -- stopping pipeline, transitioning to GST_STATE_NULL
[gstreamer] gstCamera -- pipeline stopped

dusty-nv, 2022-02-14:
Hi @tonylau2010, are you able to run `video-viewer.py /dev/video0` ?   Can you provide me the full terminal log from it?

You might want to run detectnet.py with `--headless` flag to see if it's related to the rendering/display.

tonylau2010, 2022-02-15:
> 
@dusty-nv 
Thanks a Lot for your reply!
The log is as follows，when do this： sudo video-viewer.py /dev/video1
tony-nano@tonynano-desktop:~/Downloads/readyZip/jetson-inference/build/aarch64/bin$ sudo video-viewer.py /dev/video1
nvbuf_utils: Could not get EGL display connection
[gstreamer] initialized gstreamer, version 1.14.5.0
[gstreamer] gstCamera -- attempting to create device v4l2:///dev/video1

(python3:10444): GStreamer-CRITICAL **: 21:38:51.850: gst_element_message_full_with_details: assertion 'GST_IS_ELEMENT (element)' failed

(python3:10444): GStreamer-CRITICAL **: 21:38:51.851: gst_element_message_full_with_details: assertion 'GST_IS_ELEMENT (element)' failed
[gstreamer] gstCamera -- found v4l2 device: USB2.0 Camera
[gstreamer] v4l2-proplist, device.path=(string)/dev/video1, udev-probed=(boolean)false, device.api=(string)v4l2, v4l2.device.driver=(string)uvcvideo, v4l2.device.card=(string)"USB2.0\ Camera", v4l2.device.bus_info=(string)usb-70090000.xusb-2.3, v4l2.device.version=(uint)264701, v4l2.device.capabilities=(uint)2216689665, v4l2.device.device_caps=(uint)69206017;
[gstreamer] gstCamera -- found 5 caps for v4l2 device /dev/video1
[gstreamer] [0] video/x-raw, format=(string)YUY2, width=(int)640, height=(int)480, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)30/1;
[gstreamer] [1] video/x-raw, format=(string)YUY2, width=(int)352, height=(int)288, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)30/1;
[gstreamer] [2] video/x-raw, format=(string)YUY2, width=(int)320, height=(int)240, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)30/1;
[gstreamer] [3] video/x-raw, format=(string)YUY2, width=(int)176, height=(int)144, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)30/1;
[gstreamer] [4] video/x-raw, format=(string)YUY2, width=(int)160, height=(int)120, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)30/1;
[gstreamer] gstCamera -- selected device profile:  codec=raw format=yuyv width=640 height=480
[gstreamer] gstCamera pipeline string:
[gstreamer] v4l2src device=/dev/video1 ! video/x-raw, format=(string)YUY2, width=(int)640, height=(int)480 ! appsink name=mysink
[gstreamer] gstCamera successfully created device v4l2:///dev/video1
[video]  created gstCamera from v4l2:///dev/video1
------------------------------------------------
gstCamera video options:
------------------------------------------------
  -- URI: v4l2:///dev/video1
     - protocol:  v4l2
     - location:  /dev/video1
     - port:      1
  -- deviceType: v4l2
  -- ioType:     input
  -- codec:      raw
  -- width:      640
  -- height:     480
  -- frameRate:  30.000000
  -- bitRate:    0
  -- numBuffers: 4
  -- zeroCopy:   true
  -- flipMethod: none
  -- loop:       0
  -- rtspLatency 2000
------------------------------------------------
[OpenGL] glDisplay -- X screen 0 resolution:  1364x768
[OpenGL] glDisplay -- X window resolution:    1364x768
[OpenGL] glDisplay -- display device initialized (1364x768)
[video]  created glDisplay from display://0
------------------------------------------------
glDisplay video options:
------------------------------------------------
  -- URI: display://0
     - protocol:  display
     - location:  0
  -- deviceType: display
  -- ioType:     output
  -- codec:      raw
  -- width:      1364
  -- height:     768
  -- frameRate:  0.000000
  -- bitRate:    0
  -- numBuffers: 4
  -- zeroCopy:   true
  -- flipMethod: none
  -- loop:       0
  -- rtspLatency 2000
------------------------------------------------
[gstreamer] opening gstCamera for streaming, transitioning pipeline to GST_STATE_PLAYING
[gstreamer] gstreamer changed state from NULL to READY ==> mysink
[... 20 lines cut ...]
[gstreamer] gstreamer changed state from PAUSED to PLAYING ==> pipeline0
[OpenGL] glDisplay -- set the window size to 640x480
[OpenGL] creating 640x480 texture (GL_RGB8 format, 921600 bytes)
[cuda]      unknown error (error 999) (hex 0x3E7)
[cuda]      /home/tony-nano/Downloads/readyZip/jetson-inference/utils/display/glTexture.cpp:360
[cuda]      unknown error (error 999) (hex 0x3E7)
[cuda]      /home/tony-nano/Downloads/readyZip/jetson-inference/utils/cuda/cudaYUV-YUYV.cu:160
[cuda]      unknown error (error 999) (hex 0x3E7)
[cuda]      /home/tony-nano/Downloads/readyZip/jetson-inference/utils/cuda/cudaColorspace.cpp:75
[cuda]      unknown error (error 999) (hex 0x3E7)
[cuda]      /home/tony-nano/Downloads/readyZip/jetson-inference/utils/codec/gstBufferManager.cpp:383
[gstreamer] gstBufferManager -- unsupported image format (rgb8)
[gstreamer]                     supported formats are:
[gstreamer]                        * rgb8
[gstreamer]                        * rgba8
[gstreamer]                        * rgb32f
[gstreamer]                        * rgba32f
[gstreamer] gstDecoder -- failed to retrieve next image buffer
Traceback (most recent call last):
  File "/usr/local/bin/video-viewer.py", line 51, in <module>
    image = input.Capture()
Exception: jetson.utils -- videoSource failed to capture image
[gstreamer] gstCamera -- stopping pipeline, transitioning to GST_STATE_NULL
[gstreamer] gstCamera -- pipeline stopped
tony-nano@tonynano-desktop:~/Downloads/readyZip/jetson-inference/build/aarch64/bin$

tonylau2010, 2022-02-15:
Reopen now,the problem is still occur

dusty-nv, 2022-02-15:
Hmm I'm not entirely sure, what's the camera you are using?

Do you have a different camera you could try?

tonylau2010, 2022-02-16:
Thanks a lot,camera is not supported，changeing one is working.The problem is solved！ @dusty-nv

closed 2022-02-16 by tonylau2010