dusty-nv/jetson-inference #1363: YOLO3 onnx model

karimhussein67, 2022-02-17:
I trained custom image detection model via Yolo3 algorithm, then I converted it to onnx and trt.
I'm trying to run the model  with:

`yolo3_net = jetson.inference.detectNet(argv=['--model=models/knife/yolov3-416.onnx', '--labels=models/knife/labels.txt', '--input-blob=input0', '--output-cvg=scores', '--output-bbox=boxes'] , threshold= 0.5)`

But I got this error:

> [TRT]    3: Cannot find binding of given name: input0
> [TRT]    failed to find requested input layer input0 in network
> [TRT]    device GPU, failed to create resources for CUDA engine
> [TRT]    failed to create TensorRT engine for models/knife/yolov3-416.onnx, device GPU
> [TRT]    detectNet -- failed to initialize.
> jetson.inference -- detectNet failed to load network
> Traceback (most recent call last):
>   File "face-recognition-module/stream.py", line 132, in <module>
>     '--input-blob=input0', '--output-cvg=scores', '--output-bbox=boxes'] , threshold= 0.5)
> Exception: jetson.inference -- detectNet failed to load network

I have one class in my output layer, so I don't know what parameters should I put.
Thank you

dusty-nv, 2022-02-17:
Hi @karimhussein67, for detectNet ONNX, this repo is setup for SSD-Mobilenet ONNX trained with PyTorch.   Adapting it for YOLO would require changing the pre/post-processing.

To run YOLOv3 with TensorRT, you may refer to this project:  https://jkjung-avt.github.io/tensorrt-yolov3/

closed 2022-02-21 by karimhussein67

Ali-Raza12, 2023-02-15:
> Hi @karimhussein67, for detectNet ONNX, this repo is setup for SSD-Mobilenet ONNX trained with PyTorch. Adapting it for YOLO would require changing the pre/post-processing.
> 
> To run YOLOv3 with TensorRT, you may refer to this project: https://jkjung-avt.github.io/tensorrt-yolov3/

Hi @dusty-nv , what if we create tensorRT engine without using jetson-inference repository. Do we still need to do any further changes. Actually I have converted Yolov7.pt to .onnx and then passed the file to jetson.inference.detectNet(). It created the trt engine successfully but I am not getting any detections upon inferencing. Also it is not giving any sort of error.