dusty-nv/jetson-inference #1367: ImageNet failed to load network

eanmikale, 2022-02-18:
Dusty, 
recently trained a network on the Jetson Xavier AGX, pointing all data to my SD Card. I am now running the model on the Nano, using the same card for rapid prototyping. However, I am getting the following error, that the forums have not been able to help me solve: 

[TRT]    imageNet -- loaded 280 class info entries
[TRT]    imageNet -- didn't load expected number of class descriptions  (280 of 123)
[TRT]    imageNet -- failed to load synset class descriptions  (280 / 280 of 123)
[TRT]    imageNet -- failed to initialize.
jetson.inference -- imageNet failed to load built-in network 'googlenet'
Traceback (most recent call last):
  File "/usr/local/bin/imagenet.py", line 55, in <module>
    net = jetson.inference.imageNet(opt.network, sys.argv)
Exception: jetson.inference -- imageNet failed to load network

I searched through the data and network directories, as well as with the synset file and could not find anything corrupt, or erroneous. 

Maybe using the last save checkpoint? I exported the file on a different machine. Does that matter. It took 3 days to train. Please don't tell me I need to retrain and immediately export? Any assistance would be great.

dusty-nv, 2022-02-18:
Hi @eanmikale, what is the command you are using to run your model?

It looks like the labels.txt has more entries than expected - your labels.txt has 280 classes, while the model has 123 classes.

eanmikale, 2022-02-19:
Dusty,I got it to run. I used the checkpoint as opposed to best model, which the chechpoint had proper classes. This solved my issue. Hope it helps others.  ---- On Fri, 18 Feb 2022 11:33:45 -0600  ***@***.***  wrote ----
Hi @eanmikale, what is the command you are using to run your model?
It looks like the labels.txt has more entries than expected - your labels.txt has 280 classes, while the model has 123 classes.

closed 2023-04-27 by dusty-nv